mavlink/qgroundcontrol #9688: QGroundControl on first launch always starts with indoorPalette

akontsevich, 2021-06-01:
## Expected Behavior
QGC should start with `indoorPalette=0` (light mode) setting on first start as it set as the default in `App.SettingsGroup.json` file, ...

## Current Behavior
... however it always starts in dark mode (`indoorPalette=1`).

## Steps to Reproduce:
1. Clean up `~/.config/QGroundControl.org/QGroundControl Daily.ini` file
2. Start fresh daily build of QGroundControl
3. Started in dark mode while light is the default:

```
{
    "name":             "indoorPalette",
    "shortDesc": "Application color scheme",
    "longDesc":  "The color scheme for the user interface.",
    "type":             "uint32",
    "enumStrings":      "Indoor,Outdoor",
    "enumValues":       "1,0",
    "default":     0
},
```
## System Information
- Operating System: Ubuntu 20.04
- QGC Version: 2021-05-29
- QGC build: daily

## Log Files and Screenshots
Settings on the first start:

[image]

DonLakeFlyer, 2021-06-04:
I don't think that initial value is respected. If I remember desktop builds start with indoor and mobile builds start with outdoor. Whether that continues to make any sense is debatable.

akontsevich, 2021-06-04:
We wanted to control this parameter, however our fork always started with indoor, while outdoor is default. Forced to enable it explicitly. So definitely default `Fact` logic works incorrect (it is very strange initial value is not respected) which is confusing and sounds like a bug.

closed 2024-05-31 by HTRamsey