mavlink/qgroundcontrol #9692: Use of custom-extention seems to break debug and other builds

rworrall, 2021-06-04:
## Expected Behavior
QGC (master) compiles (including custom / custom-exension) variants:

## Current Behavior

Builds fail when utilizing "custom extension"

Compiler error (Lin/Win):

c:\users\rtw\desktop\qgroundcontrol\src\missionmanager\MissionItem.h(43): error C2061: syntax error: identifier 'MAV_CMD'
c:\users\rtw\desktop\qgroundcontrol\src\missionmanager\MissionItem.h(63): error C3646: 'command': unknown override specifier
c:\users\rtw\desktop\qgroundcontrol\src\missionmanager\MissionItem.h(63): error C2059: syntax error: '('

## Steps to Reproduce:

1. git clone https://github.com/mavlink/qgroundcontrol.git --recursive
2. cd qgroundcontrol
3. touch user_config.pri
4. mv custom-example custom
5. custom/updateqrc.py
6. compile...

## System Information
When posting bug reports, include the following information
- Operating System:Windows 10 x86_64 / Ubuntu 20.04 x86_64
- QGC Version: [master]
- QGC build: [master]
- Flight Controller: N/A
- Autopilot: N/A

## Detailed Description

[image]

[image]

DonLakeFlyer, 2021-06-04:
Are you sure you have the mavlink submodule updated?

rworrall, 2021-06-04:
I think so.... The issue happens just when using custom-extension --> custom.  New git clone source tree works until steps 4 and 5 above (mv custom-example custom; custom/updateqrc.py).  Non-custom builds compile for ADR/LIN/WIN, custom module introduced (as in how to reproduce) and wheels come off during build...?

 aef704108f3c1e7520338c34171c28565211a607 libs/mavlink/include/mavlink/v2.0 (aef70410)

rworrall, 2021-06-22:
Steps to Reproduce:

    git clone https://github.com/mavlink/qgroundcontrol.git --recursive
    cd qgroundcontrol
    touch user_config.pri
    mv custom-example custom
    custom/updateqrc.py
    compile...

Not able to **compile** on any platform / for any target after above steps.

jhw84, 2021-06-23:
I have the same issue. Builds fine without the custom example, but when enabling it (following the steps in the readme) the build fails because MAV_CMD is not declared in MissionItem.h:

`/home/me/qgroundcontrol/src/MissionManager/MissionItem.h:45: error: ‘MAV_CMD’ has not been declared
In file included from [...]`

Mavlink submodule is up to date and other Mavlink types such as MAV_FRAME in the same file do not appear to be causing any problems.

rworrall, 2021-06-24:
Thank you for the comment / work-around!

DonLakeFlyer, 2021-06-24:
> touch user_config.pri

This is  most likely what is causing the problem. That kicks off a bug in the build scripts. You don't need to do this. You shouldn't have an empty user_config.pri

DonLakeFlyer, 2021-06-24:
No, that's not it. Still looking.

DonLakeFlyer, 2021-06-24:
The mavlink C generator must have had a bug in it at some point which generated bad headers. The mavlink headers submodule which is currently on master is bad. Newer ones are correct. I'm working to update the submodule.

DonLakeFlyer, 2021-06-24:
Latest mavlink headers no longer have bug

closed 2021-06-24 by DonLakeFlyer

rworrall, 2021-06-26:
Noted with respect to user_config.pri   - Thank you.

PR #9723 (2021-06-24): Mavlink update
Fix #9692