mavlink/qgroundcontrol #9715: GPU Compatibility Mode in QGC is not working.

WTPENGUIN, 2021-06-17:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
QGC run normaly.

## Current Behavior
The QGC screen does not appear.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Install QGC
2. Click QGroundControl (GPU Compatibility Mode)

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: 4.1.3
- QGC build: stable
- Flight Controller:
- Autopilot (with version):

## Detailed Description
When running in GPU Compatibility mode, the process was found to be running, although there was no response.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

booo, 2021-10-10:
@WTPENGUIN could you please cleanup the bug report and add a bit more information? Can you start QGC from the commandline and check if there is any output you can paste here?

DonLakeFlyer, 2021-11-08:
Same thing seems to happen with Stable so hasn't been working for a while