mavlink/qgroundcontrol #9729: Build failure: no member named 'microhardManager' in 'QGCToolbox'

phma, 2021-06-29:
I'm trying to build QGC (haven't succeeded yet) and getting this error:
```
In file included from /home/phma/src/qgroundcontrol/src/Microhard/MicrohardSettings.cc:13:
In file included from /home/phma/src/qgroundcontrol/src/QGCApplication.h:27:
In file included from /home/phma/src/qgroundcontrol/src/FirmwarePlugin/FirmwarePluginManager.h:14:
/home/phma/src/qgroundcontrol/src/FirmwarePlugin/FirmwarePlugin.h:334:44: warning: unused parameter 'vehicle' [-Wunused-parameter]
    virtual QString getHobbsMeter(Vehicle* vehicle) { return "Not Supported"; }
                                           ^
/home/phma/src/qgroundcontrol/src/Microhard/MicrohardSettings.cc:83:53: error: no member named 'microhardManager' in 'QGCToolbox'
        std::string userName = qgcApp()->toolbox()->microhardManager()->configUserName().toStdString() + "\n";
                               ~~~~~~~~~~~~~~~~~~~  ^
/home/phma/src/qgroundcontrol/src/Microhard/MicrohardSettings.cc:86:48: error: no member named 'microhardManager' in 'QGCToolbox'
        std::string pwd = qgcApp()->toolbox()->microhardManager()->configPassword().toStdString() + "\n";
                          ~~~~~~~~~~~~~~~~~~~  ^
/home/phma/src/qgroundcontrol/src/Microhard/MicrohardSettings.cc:92:34: error: no member named 'microhardManager' in 'QGCToolbox'
            qgcApp()->toolbox()->microhardManager()->setEncryptionKey();
            ~~~~~~~~~~~~~~~~~~~  ^
1 warning and 3 errors generated.
ninja: build stopped: subcommand failed.
```
## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu Focal
- QGC Version: Just pulled from git
- QGC build: not successfully built yet
- Flight Controller: n/a
- Autopilot (with version): n/a
- CMake: 3.16.3
- Ninja: 1.10.0
- Clang: 10.0.0-4ubuntu1

DonLakeFlyer, 2021-06-29:
Looks like you turned on microhard support. I don't think that is working.

closed 2021-06-29 by DonLakeFlyer

phma, 2021-06-29:
I haven't turned on or off anything. I just pulled the latest and tried to build it.

What is microhard, and how do I turn it off?

patrickelectric, 2021-06-30:
@DonLakeFlyer I found the same problem when building from scratch today using master branch, it appears that there is a problem with the microhard configuration where is enabled by default

DonLakeFlyer, 2021-07-01:
Wait, you are using cmake. qmake is working fine. cmake is likely broken. microhard should be off.

patrickelectric, 2021-07-01:
> qmake is working fine. cmake is likely broken

CMake enables both taisync and microhard by default when gst is found.

phma, 2021-07-01:
Yes, I'm using cmake. Can we reopen this? My earlier problem with using cmake (libevents wasn't mentioned in CMakeLists) was fixed by a recent commit.

I've been using cmake for many years for my projects, but don't know qmake and haven't figured out how to get debug/release/fuzzing builds, how to choose the compiler, how to generate Ninja files, or how to point it to ~/Qt/5.12.6 instead of the Qt installed by dpkg. I can do all this easily with cmake.

patrickelectric, 2021-07-01:
@phma I should create a PR to fix it today.

patrickelectric, 2021-07-01:
Check https://github.com/mavlink/qgroundcontrol/pull/9733

phma, 2021-07-01:
I got "You are not authorized to merge this pull request", so I edited the file by hand. The options appear, turned off, in CMakeCache.txt, but I still get the same compile error. I left a comment on the pull request.

phma, 2021-07-01:
I made one more change and got it to build. I'm going to submit a pull request.

phma, 2021-07-01:
https://github.com/mavlink/qgroundcontrol/pull/9736

PR #9733 (2021-07-01): cmake: Turn MICROHARD and TAISYNC disabled by default
Fix #9729