mavlink/qgroundcontrol #9741: No elevation data for Norway?

AndKe, 2021-07-02:
## Expected Behavior
expected to see elevation

## Current Behavior
errors, no elevation data

## Steps to Reproduce:
Monitor a mission in northern Norway
```
./qgroundcontrol-start.sh 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Settings location "/home/andre/.config/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\n"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
PX4ParameterMetaDataLog: Invalid max value, name: "TRIG_PINS_EX"  type: 5  max: "4294967040"  error: "Value must be within 0 and 2147483647"
serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT.
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/andre/Documents/QGroundControl Daily/Logs"
Map Cache in: "/home/andre/.cache/QGCMapCache300" / "qgcMapCache.db"
PX4ParameterMetaDataLog: Invalid max value, name: "TRIG_PINS_EX"  type: 5  max: "4294967040"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid max value, name: "TRIG_PINS_EX"  type: 5  max: "4294967040"  error: "Value must be within 0 and 2147483647"
qml: QGCCorePlugin(0x556f5dfb0f50) []
<Unknown File>: Invalid empty URL
Adding target QHostAddress("127.0.0.1") 51691
PX4ParameterMetaDataLog: Invalid max value, name: "TRIG_PINS_EX"  type: 5  max: "4294967040"  error: "Value must be within 0 and 2147483647"
libpng warning: iCCP: known incorrect sRGB profile
TerrainQueryLog: Error in fetching elevation tile. Empty response.
qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000"
QGCPositionManager error 0
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.
TerrainQueryLog: Error in fetching elevation tile. Empty response.

```

## System Information
When posting bug reports, include the following information
Ubuntu 21.04
QGC version: today's HEAD
ArduPlane in SITL

DonLakeFlyer, 2021-07-07:
If I remember correctly there is no elevation data above/below a certain latitude. Don't remember what it is

AndKe, 2021-07-16:
I see no mention of such limitation on https://developers.google.com/maps/documentation/elevation/overview - Also, I had no problems using this feature before.

Please note that I get this error regardless of where I put a waypoint, even in central Europe.

DonLakeFlyer, 2021-07-16:
> Please note that I get this error regardless of where I put a waypoint, even in central Europe.

Ah, that's a problem then. QGC doesn't use Google, elevation comes from AirMap.

AndKe, 2021-07-16:
is that service also feeding TERRAIN_DATA / TERRAIN_REQUEST to ArduPilot ?
I hope you can add google, and even an option to add own SRTM files.

DonLakeFlyer, 2021-07-16:
> is that service also feeding TERRAIN_DATA / TERRAIN_REQUEST to ArduPilot ?

Not that I know of. As far as I know MP pulls from an adrupilot server which has SRTM tiles on it which came from x-plane I think.

DonLakeFlyer, 2021-07-16:
> is that service also feeding TERRAIN_DATA / TERRAIN_REQUEST to ArduPilot ?

Wait may have the question wrong. If you mean does QGC feed airmap elevation data to the mavlink terrain protocol. The answer is yes. The only place QGC gets terrain data from is airmap.

AndKe, 2021-07-16:
ok, this requires a serious change then - also, I now know why somebody told me that QGC did not provide terrain data, while I answered that I think it does...

DonLakeFlyer, 2021-07-16:
> I now know why somebody told me that QGC did not provide terrain data

Huh? QGC works fine with ArduPilot mavlink terrain tile protocol support. That is what is was implemented to and tested on ages ago. PX4 doesn't even support it yet.

DonLakeFlyer, 2021-07-16:
> QGC works fine

Or at least it will once I look at this bug. But this is the first report I have of terrain data not downloading. I just did a large change to support terrain follow TERRAIN_FRAME pattern generation on Surveys using mavlink tile protocol. All working fine when I implemented that. That could have been what someone was talking about.

DonLakeFlyer, 2021-07-20:
> Please note that I get this error regardless of where I put a waypoint, even in central Europe.

@AndKe This seems to be working fine for me. Can you turn on TerrainQueryLog and post the output?

AndKe, 2021-07-20:
I am going on vacation about right now, but found time for this -- please see attached log.

[elevation.zip](https://github.com/mavlink/qgroundcontrol/files/6850780/elevation.zip)
[image]

Ryanf55, 2024-04-06:
Same as https://github.com/mavlink/qgroundcontrol/issues/10592

DonLakeFlyer, 2024-04-06:
@AndKe Does the new elevation server have Norway data?

AndKe, 2024-04-06:
@DonLakeFlyer Yes, and even the northmost parts of Svalbard.

AndKe, 2024-04-06:
And "Thank you"

closed 2024-04-06 by AndKe

DonLakeFlyer, 2024-04-06:
Great news then!