mavlink/qgroundcontrol #9756: Altitude visualization fail.

AndKe, 2021-07-08:
## Expected Behavior
seeing mission altitudes in the graph on the bottom of the screen

## Current Behavior
[image]

## Steps to Reproduce:
load mission:
```
QGC WPL 110
0	1	0	16	0	0	0	0	69.040305	23.034362	363.630000	1
1	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.03247370	23.03043370	120.000000	1
2	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.03314940	23.01549910	120.000000	1
3	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.04718060	23.02133560	120.000000	1
4	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.04656670	23.03369520	120.000000	1
5	0	0	177	1.00000000	-1.00000000	0.00000000	0.00000000	0.00000000	0.00000000	0.000000	1
6	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.05098620	23.02373890	80.000000	1
7	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.05110900	23.03378110	50.000000	1
8	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.04866010	23.03705330	30.000000	1
9	0	0	178	0.00000000	32.00000000	0.00000000	0.00000000	0.00000000	0.00000000	0.000000	1
10	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.04360720	23.03525330	24.000000	1
11	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.04099780	23.03409720	26.000000	1
12	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.03731010	23.03258780	29.000000	1
13	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.03482230	23.03149310	80.000000	1
14	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.03311100	23.02208660	80.000000	1
15	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.03195160	23.02554130	50.000000	1
16	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.03250210	23.03041760	30.000000	1
17	0	0	178	0.00000000	32.00000000	0.00000000	0.00000000	0.00000000	0.00000000	0.000000	1
18	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.03730510	23.03269140	29.000000	1
19	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.04099540	23.03416560	26.000000	1
20	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.04360410	23.03533610	24.000000	1
21	0	3	16	0.00000000	0.00000000	0.00000000	0.00000000	69.04864850	23.03717670	80.000000	1
22	0	0	177	6.00000000	-1.00000000	0.00000000	0.00000000	0.00000000	0.00000000	0.000000	1
```

DonLakeFlyer, 2021-07-20:
This is a dup of #9741 isn't it?

AndKe, 2021-07-21:
Yes, while waiting, I must have forgotten to have reported it earlier.