mavlink/qgroundcontrol #9763: QGC is failing to build with

patrickelectric, 2021-07-12:
The Following PR should fix the issue, but it needs to be merged:
https://github.com/mavlink/gst-plugins-good/pull/1

## Expected Behavior
QGC should compile

## Current Behavior
Failing to compile

## Steps to Reproduce:
1. Install dependencies
2. Use the latest glib2 version available (2.68.3-1)

## System Information
- Operating System: Arch
- QGC Version: master (cb3bc2615020e96fc88e764f76d310c2d69a696e)
- QGC build: master

## Detailed Description
Fails with both QMake and CMake build system

## Log Files and Screenshots

<details>
  <summary>CMake</summary>
```
[ 11%] Built target VehicleSetup_autogen                                                                                                                                                                                                                      
In file included from /usr/include/glib-2.0/glib/gthread.h:32,                                                                                                                                                                                                
                 from /usr/include/glib-2.0/glib/gasyncqueue.h:32,                                                             
                 from /usr/include/glib-2.0/glib.h:32,                                                                         
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,                                                                                                                                                                                                
                 from /home/patrick/git/blue/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.h:24,                                                                                                                                       
                 from /home/patrick/git/blue/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc:25:                                                                                                                                      
/home/patrick/git/blue/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc: In function ‘GstGLDisplay* gst_qt_get_gl_display()’:                                                                                                          
/usr/include/glib-2.0/glib/gatomic.h:117:19: error: argument 2 of ‘__atomic_load’ must not be a pointer to a ‘volatile’ type                                                                                                                                  
  117 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \                                           
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                             
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro ‘g_atomic_pointer_get’                                                                                                                                                                
  260 |     (!g_atomic_pointer_get (location) &&                             \                                                                                                                                                                                
      |       ^~~~~~~~~~~~~~~~~~~~                                                                                             
/home/patrick/git/blue/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc:65:7: note: in expansion of macro ‘g_once_init_enter’                                                                                                          
   65 |   if (g_once_init_enter (&_debug)) {                                                                                                                                                                                                                  
      |       ^~~~~~~~~~~~~~~~~                                                                                                
make[2]: *** [libs/qmlglsink/CMakeFiles/qmlglsink.dir/build.make:104: libs/qmlglsink/CMakeFiles/qmlglsink.dir/gst-plugins-good/ext/qt/gstqtglutility.cc.o] Error 1                                                                                            
make[2]: *** Waiting for unfinished jobs....                                                                                   
[ 11%] Built target QmlControls_autogen                                                                                        
[ 11%] Built target Settings_autogen                                                                                                                                                                                                                          
[ 11%] Built target FirmwarePlugin_autogen                                                                                     
In file included from /usr/include/glib-2.0/glib/gthread.h:32,                                                                 
                 from /usr/include/glib-2.0/glib/gasyncqueue.h:32,                                                             
                 from /usr/include/glib-2.0/glib.h:32,                                                                         
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,                                                                 
                 from /usr/include/gstreamer-1.0/gst/video/video.h:23,                                                         
                 from /home/patrick/git/blue/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc:28:                                                                                                                                       
/home/patrick/git/blue/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc: In constructor ‘GstQSGTexture::GstQSGTexture()’:                                                                                                               
/usr/include/glib-2.0/glib/gatomic.h:117:19: error: argument 2 of ‘__atomic_load’ must not be a pointer to a ‘volatile’ type                                                                                                                                  
  117 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \                                           
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                              
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro ‘g_atomic_pointer_get’                                                                                                                                                                
  260 |     (!g_atomic_pointer_get (location) &&                             \                                                 
      |       ^~~~~~~~~~~~~~~~~~~~                             
/home/patrick/git/blue/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc:42:7: note: in expansion of macro ‘g_once_init_enter’                                                                                                           
   42 |   if (g_once_init_enter (&_debug)) {                   
      |       ^~~~~~~~~~~~~~~~~                                
make[2]: *** [libs/qmlglsink/CMakeFiles/qmlglsink.dir/build.make:118: libs/qmlglsink/CMakeFiles/qmlglsink.dir/gst-plugins-good/ext/qt/gstqsgtexture.cc.o] Error 1                                                                                             
In file included from /usr/include/glib-2.0/glib/gthread.h:32,                                                                 
                 from /usr/include/glib-2.0/glib/gasyncqueue.h:32,                                                             
                 from /usr/include/glib-2.0/glib.h:32,                                                                         
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,                                                                 
[... cut ...]
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro ‘g_atomic_pointer_get’
  260 |     (!g_atomic_pointer_get (location) &&                             \
      |       ^~~~~~~~~~~~~~~~~~~~
/home/patrick/git/blue/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/qtwindow.cc:160:7: note: in expansion of macro ‘g_once_init_enter’
  160 |   if (g_once_init_enter(&once)) {
      |       ^~~~~~~~~~~~~~~~~
make: *** [Makefile:150945: qtwindow.o] Error 1
```
</details>

jaxxzer, 2021-08-04:
duplicate https://github.com/mavlink/qgroundcontrol/issues/9669?

patrickelectric, 2021-08-04:
Yep!

closed 2021-08-04 by patrickelectric