mavlink/qgroundcontrol #9768: which one is the offical release version of which I could used for redevelopment?

lukezhqin, 2021-07-15:
Tell us a bit about the feature:
Hi, I'd like to know which branch is the offical branch which should I used for redevelopment?
I download exe file from [QGC](http://qgroundcontrol.com/)-DOWNLOADS-WINDOWS. And I found it contains fuctions in branch Stable_V4.1 but not the same with the release tag such as v4.1.3. Also I found that  [QGC](http://qgroundcontrol.com/)-DOWNLOADS-DAILY BUILDS just like the release tag v4.1.3.

Any additional context you can provide will make the feature easier to evaluate (e.g. mockups, detailed specification, etc.)

holden-zenith, 2022-02-11:
This should be closed

bkueng, 2022-02-15:
It's generally the `Stable_` branches.

closed 2022-02-15 by bkueng