mavlink/qgroundcontrol #9770: SENS_GPS_MASK bits do not match PX4 definition

petertorelli, 2021-07-15:
## Expected Behavior
When modifying bits in the SENS_GPS_MASK parameter under Vehicle > Parameters > Sensors, the SENS_GPS_MASK bits' toggle inputs do not reflect the current definition from the PX4 project documentation [for the SENS_GPS_MASK parameter](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#SENS_GPS_MASK).

## Current Behavior
Bit 0 is speed, Bit 1 is hpos, bit 2 is vpos. If I set all three toggles, I should get 7, instead I get 3. Toggling the checkbox for speed does nothing, `vpos` affects bit 0, and `hpos` affects bit 1. `hpos` is the only correct toggle.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Connect a PX4 flight controller
2. Go to Vehicle > Parameters > Sensors
3. Attempt to manipulate the toggles as mentioned above
4. Observe the incorrect results

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: development HEAD: 78cf9bbe6 2021-05-06
- QGC build: see above
- Flight Controller: PixHawk4 Mini 1.12
- Autopilot (with version): PX4 1.12

## Detailed Description
That's pretty much it.

## Log Files and Screenshots
In this screenshot, speed is selected (bit 0) but the value is still 0, and it should be 1.

[image]

MaEtUgR, 2021-11-15:
@petertorelli This sounds exactly like an issue that was fixed very recently: https://github.com/mavlink/qgroundcontrol/issues/9854

The issue is I can't get a hold of the daily version to test and verify because CI doesn't provide it anymore at the moment https://github.com/mavlink/qgroundcontrol/issues/9946 🤔

Let's reopen if we see the issue again with the latest version (once we get it).

closed 2021-11-15 by MaEtUgR

petertorelli, 2021-11-15:
@MaEtUgR Thanks!