mavlink/qgroundcontrol #9771: No such file or directory

zijian98, 2021-07-16:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Build qgroundcontrol using Qt

## Current Behavior
airmap/qt/client.h: No such file or directory

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Follow https://dev.qgroundcontrol.com/master/en/getting_started/
2. Install Qt in default location
3.Attempt to build with Qt creator
4. airmap/qt/client.h: No such file or directory

Any ways to fix?

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04.2.0 LTS

patrickelectric, 2021-08-02:
`git submodule update --init --recursive` should fix the problem.

bitsauce, 2021-12-06:
Hi, I just encountered the same error earlier today. In my case, I found out that this was caused by the fact that the Airmap SDK is supposed to be automatically downloaded via `curl` (on Ubuntu) at build-time, however, `curl` was not installed on my machine which meant that these SDK files didn't get downloaded properly. Installing `curl` with `sudo apt install curl` and doing a rebuild of QGC fixed this issue for me

jingwenyi, 2022-01-06:
you should  build in terminal
like this:
mkdir build
cd build
qmake ../
make -j4

Manodiestra, 2022-02-15:
I'm having the same issue on Ubuntu 20.04 as well. Except that mine says that the file is (not found) in
`airmap/client.h` instead of `airmap/qt/client.h` as mentioned by the OP.
I've tried all of the fixes mentioned here, except for building in the terminal, which doesn't work for me. I get an error `qmake: could not find a Qt installation of ''` with the qmake terminal command `qmake ../`.
I installed Qt using the qt online installer as mentioned in the [QGC Developer Guide](https://dev.qgroundcontrol.com/master/en/getting_started/) and am trying to build with Qt Creator.

enetkilifatih, 2022-03-07:
I got the same error too. And I found what the problem is, however I couldn't introduce a solution because of lack of qmake knowledge.
If you don't use `-j` param (no parallel build), it just builds without any error. Project depends the airmap to be downloaded, however it doesn't wait to complete the download during parallel build.
When you get that error (with successful download of airmap sdk), rebuilding without cleaning the build dir will be successful. You may suppose that something you did fixes the issue. If you do clean cache and rebuild then you will get the same error again until someone fixes that dependency issue.

Someone have faced a similar problem here.
https://stackoverflow.com/questions/50584019/pre-targetdeps-fails-when-using-parallel-builds

closed 2024-09-22 by HTRamsey