mavlink/qgroundcontrol #9781: QGC rounding Float Parameters

LowOrbitIonCannon, 2021-07-23:
## Expected Behavior
QGC should show Float Parameters with decimal Places, even if there is no rule given for handling the decimal places.

## Current Behavior
QGC shows Custom Parameters rounded to the nearest Integer.
FMU receives the correct Parameter in FLOAT. Also QGC recognizes that the Parameters on the FMU are not Rounded.
[image]

Standard Parameters show correct behavior:
[image]

## Steps to Reproduce:
1. Introduce Custom float Parameter
2. Set value to something with decimal places

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: [e.g. 3.4.4]
- QGC build: HEAD:78cf9bbe6 
- Flight Controller: nxp_fmuk66-v3
- https://github.com/PX4/PX4-Autopilot/pull/14474

## Detailed Description
I added custom YAML Parameters. When I change them in QGC, they'll be correctly set in the FMU but won't display properly.
adding decimal Metadata fixes the Display.

## Log Files and Screenshots
Mavlink Console snapshot showing that the Parameters are correct on the FMU:
[image]

@DonLakeFlyer

jhiggason, 2023-02-20:
I can confirm this bug and it caused me HUGE problems when setting up a drone because I didn't understand that even though I set a parameter...it wasn't displaying it properly...even though it was set properly.  This needs to be fixed ASAP.

DonLakeFlyer, 2023-02-20:
Create a PX4 branch somewhere based on master which adds a custom parameter which fails like this that I can then compile and run in SITL and then I can try to fix this.

jhiggason, 2023-02-20:
@DonLakeFlyer Not sure I 100% understand this- you just want me to branch the master?  I was using the Daily build of QGC and Master of PX4. (As of like 2 weeks ago)   Lots of Avoidance settings in QGC are what are not displaying properly.  That and Ark Flow sensor settings.  Both of these came from the Master build of PX4....

DonLakeFlyer, 2023-02-20:
Ok, so this bug talks about custom parameters, not standard PX4 parameters. If this happens on them as well then tell me a specific parameter which exists in regular px4 master which if I connect QGC to a SITL instance of that will show up in the parameter list. If it requires a specific SITL vehicle to test with let me know what that is.

jhiggason, 2023-02-20:
cp_dist is one.  Rounds to whole numbers even though it allows for float.  And again, when setting it in QGC- it works- but then it just doesn't display it properly.  Like the field to view the parameter in QGC is dec instead of float. There are lots though that I ran into while working with it.  Do you think we can just compare QGC and PX4 parameter field values to ensure they are all matching?

jhiggason, 2023-02-21:
You are the best for looking at this, Thank you for your hard work on it!

PR #10589 (2023-03-09): FactMetaData: Fix default for decimal places
Fix #9781

When fact metadata was loaded from json if decimalPlaces was not specified it was defaulting to 0 instead of unknown. Unknown in turn defaults to 3 decimal places.
closed 2023-03-09 by mrpollo