mavlink/qgroundcontrol #9792: Video record of udp h264 video stream does not work.

kressinluiz, 2021-07-28:
## Expected Behavior
(ANDROID) When receiving a UDP H264 video stream, the video record button on PhotoVideoControl should start recording.

## Current Behavior
(ANDROID) The video record button it's not working.
(UBUNTU) It works.

## Steps to Reproduce:
1. send a udp h264 stream to qgroundcontrol with gstreamer command line tool:
`gst-launch-1.0 --gst-debug 1 videotestsrc pattern=ball -v ! x264enc ! rtph264pay ! udpsink host=192.168.2.101 port=5600`
2. open qgroundcontrol on android, go to general settings and set video source to "UDP h.264 Video Stream"
3. Go to fly view and click the record button in PhotoVideoControl.

## System Information
- Operating System: Android 9
- QGC Version: master, Stable_V4.1
- QGC build: daily
- Flight Controller: unrelated
- Autopilot (with version): unrelated

## Detailed Description
When the record button is pressed, we got the following error:
[image]

kressinluiz, 2021-07-28:
Probably related to this [issue](https://github.com/mavlink/qgroundcontrol/issues/8687) that was closed due to a confirmed fix with this [commit](https://github.com/mavlink/qgroundcontrol/commit/6ccc7941148f79305c0c9467c08b5d7fdeffbbb8).

akontsevich, 2021-08-18:
Any info when this planned to fix?

closed 2024-06-02 by HTRamsey