mavlink/qgroundcontrol #9809: Linux stable download points to a development (master) build

jaxxzer, 2021-08-04:
https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html

Linux downloads a build from S3 with the commit ash 78cf9bb
[image]

This commit is not on any stable branch/tag:
```
[jacob@jacob-desktop qgroundcontrol]$ git tag --contains 78cf9bb
refs/pull/9561/merge
[jacob@jacob-desktop qgroundcontrol]$ git branch --contains 78cf9bb
  master
```

jaxxzer, 2021-08-09:
[Windows] see also https://discuss.px4.io/t/release-version-use-strange-git-version/23665

mrpollo, 2021-08-09:
AFAIK all of the versions hosted on S3 are out of date since a few months ago. We are not sure why but GitHub Actions doesn't want to upload to S3 using the AWS CLI. The S3 upload step on CI was disabled to be able to pass tests.