mavlink/qgroundcontrol #9810: Github releases: no linux appimage or mac dmg available for 4.1.x

jaxxzer, 2021-08-04:
Where did the deployments go? https://github.com/mavlink/qgroundcontrol/releases

Related: https://github.com/mavlink/qgroundcontrol/issues/8520

mrpollo, 2021-08-09:
A quick fix, while the S3 upload issue is resolved, is to download from GitHub Actions.

You can download from the link below for macOS, choose the latest building PR (make sure it's green), click on it, and search for the "Build Artifacts" you should see an installer. Builds are also available for Linux and Windows.

https://github.com/mavlink/qgroundcontrol/actions/workflows/macos_release.yml

patrickelectric, 2021-08-09:
@mrpollo The problem is that actions got merged in master and not for 4.1, so it's not available there.
Something that we could try is to move actions to 4.1 also.

patrickelectric, 2021-08-09:
@mrpollo is it possible to get s3 fixed ? Can we get it fixed somehow ? Could we move it to BlueRobotics s3 as a temporary fix ?

mrpollo, 2021-08-09:
I have access to S3, but I don't have access to the keys for the QGC user used on GH Actions. I can try to generate a new key pair and test them locally and update GH, but I'm not sure if changing the keys will have any unintended consequences.