mavlink/qgroundcontrol #9819: /bin/sh: 1: /tmp/qt/install-getting a permission denied error qt-linux.sh: Permission denied The command '/bin/sh -c /tmp/qt/install-qt-linux.sh' returned a non-zero code: 126

prashanthravi19, 2021-08-11:
Hi i am trying to build qgroundcontrol in linux enviroment when i try to create a container I am getting following error \

/bin/sh: 1: /tmp/qt/install-qt-linux.sh: Permission denied
The command '/bin/sh -c /tmp/qt/install-qt-linux.sh' returned a non-zero code: 126

is there any solution?

patrickelectric, 2021-08-11:
Can you provide the entire command line and also output from your terminal ?

prashanthravi19, 2021-08-12:
Sending build context to Docker daemon  803.6MB
Step 1/17 : FROM ubuntu:20.04
 ---> 1318b700e415
Step 2/17 : LABEL authors="Daniel Agar <daniel@agar.ca>, Patrick José Pereira <patrickelectric@gmail.com>"
 ---> Using cache
 ---> f2092c62bdca
Step 3/17 : ARG QT_VERSION=5.12.10
 ---> Using cache
 ---> 68897d4a4bdf
Step 4/17 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> 5222e157cf1b
Step 5/17 : ENV DISPLAY :99
 ---> Using cache
 ---> b4008b46c1c0
Step 6/17 : ENV QMAKESPEC linux-g++-64
 ---> Using cache
 ---> 1fc63582ec0e
Step 7/17 : ENV QT_PATH /opt/Qt
 ---> Using cache
 ---> a9b70c68947c
Step 8/17 : ENV QT_DESKTOP $QT_PATH/${QT_VERSION}/gcc_64
 ---> Using cache
 ---> d6df90976d13
Step 9/17 : ENV PATH /usr/lib/ccache:$QT_DESKTOP/bin:$PATH
 ---> Using cache
 ---> 9762ae10e93a
Step 10/17 : RUN apt update && apt -y --quiet --no-install-recommends install  apt-utils                build-essential                 ca-certificates        ccache           checkinstall            cmake           curl            espeak fuse             g++             gcc             git             gosu           kmod             libespeak-dev           libfontconfig1          libfuse2       libgstreamer-plugins-base1.0-dev                 libgstreamer1.0-0              libgstreamer1.0-dev              libsdl2-dev             libssl-dev             libudev-dev              locales                 make            ninja-build    openssh-client           openssl                 pkg-config              rsync  speech-dispatcher                wget            xvfb            zlib1g-dev     && apt-get -y autoremove         && apt-get clean autoclean      && rm -rf /var/lib/apt/lists/{apt,dpkg,cache,log} /tmp/* /var/tmp/*
 ---> Using cache
 ---> 730ed04b3a00
Step 11/17 : COPY deploy/docker/install-qt-linux.sh root/tmp/qt/
 ---> Using cache
 ---> ae566279de9e
Step 12/17 : RUN  chmod 777 root/tmp/qt/install-qt-linux.sh
 ---> Using cache
 ---> eb4669d4f0ba
Step 13/17 : RUN locale-gen en_US.UTF-8 && dpkg-reconfigure locales
 ---> Using cache
 ---> cb5311098d9e
Step 14/17 : RUN useradd user --create-home --home-dir /home/user --shell /bin/bash --uid 1000
 ---> Using cache
 ---> 9e6c3e1db218
Step 15/17 : USER user
 ---> Using cache
 ---> 1fa4bebb41d9
Step 16/17 : WORKDIR /project/build
 ---> Using cache
 ---> ae7a7b2e5b91
Step 17/17 : CMD qmake /project/source && make -j$(nproc)
 ---> Using cache
 ---> be04c258576d
Successfully built be04c258576d
Successfully tagged qgc-linux-docker:latest
aero360@aero360dev:~/qgroundcontrolmobile/qgroundcontrol$ mkdir build

aero360@aero360dev:~/qgroundcontrolmobile/qgroundcontrol$ sudo docker run --rm -v ${PWD}:/project/source -v ${PWD}/build:/project/build qgc-linux-docker
/bin/sh: 1: qmake: not found
this is the point i am getting issue

bols-blue, 2021-11-09:
qmake cannot be found because docker does not have QT development tools.
Add the following command to the Dockerfile.
`apt install qtbase5-dev qttools5-dev-tools qt5-default`

bols-blue, 2021-11-13:
please read it.
https://dev.qgroundcontrol.com/master/en/getting_started/container.html

@booo I think this issue can be closed.

booo, 2021-11-13:
Can we add the missing command to the dockerfile?

bols-blue, 2021-11-14:
@booo He hasn't read the documentation.
It is an error that does not occur if the document is followed.

closed 2022-11-16 by booo