mavlink/qgroundcontrol #9830: MPEG-TS not playing video with decodebin3

papertiger12, 2021-08-16:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Video stream should support MPEG-TS and playback correctly.

## Current Behavior
No video is displayed.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Using the following gstream pipeline
```
gst-launch-1.0 -v videotestsrc pattern=ball \
! x264enc \
! mpegtsmux \
! rtpmp2tpay \
! udpsink host=0.0.0.0 port=5000
```
2. Open the QGC video config. Settings -> Application Settings -> Video Settings
3. Source = MPEG-TS (h.264) 
4. UDP Port = 5000
5. File format = any

## System Information
Ubuntu 20.04
QGC - LTS

## Detailed Description
Changing the decodebin version in GstVideoReceiver.cc _makeDecoder resolves the issue
```
--        if ((decoder = gst_element_factory_make("decodebin3", nullptr)) == nullptr) {
++        if ((decoder = gst_element_factory_make("decodebin", nullptr)) == nullptr) {
```

Also, using this gstreamer pipeline seems to have the same issue. 
Not working:
```
gst-launch-1.0 -v \
! udpsrc port=5000 \
! tsdemux \
! parsebin \
! decodebin3 \
! autovideosink
```

Working:
```
gst-launch-1.0 -v \
! udpsrc port=5000 \
! tsdemux \
! parsebin \
! decodebin \
! autovideosink
```

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

closed 2021-09-09 by papertiger12

hackoma, 2021-10-04:
I can confirm the same behavior in 
https://github.com/mavlink/qgroundcontrol/issues/9899

But it's work normally in v4.0.11

fredowski, 2022-09-11:
It does not work with Release Version 4.2.3 on MacOS with the above mentioned stream generation. The record button becomes red but the video window remains in the state "Waiting for Video". The following stream does not work also:

`gst-launch-1.0 -vvvvvvv videotestsrc ! 'video/x-raw, width=1920, height=1080' ! x264enc ! 'video/x-h264, profile=baseline' ! mpegtsmux ! queue ! udpsink host=127.0.0.1 port=5600` 

I installed gstreamer 1.18.6 on MacOS and I can confirm that the decodebin3 pipeline has issues:

`gst-launch-1.0 -vvvvv udpsrc port=5600 caps='video/mpegts' ! tsdemux ! parsebin ! queue ! decodebin3 ! autovideosink`

does not work while

`gst-launch-1.0 -vvvvv udpsrc port=5600 caps='video/mpegts' ! tsdemux ! parsebin ! queue ! decodebin ! autovideosink` or

`gst-launch-1.0 -vvvvv udpsrc port=5600 caps='video/mpegts' ! tsdemux ! queue ! decodebin ! autovideosink `

shows the video. As discussed in #10396 I tried to go back to decodebin but this results in "flickering" problems.