mavlink/qgroundcontrol #9839: "Communication Lost/Regained" pop up should be removed!

DavidMCampbell, 2021-08-24:
Imagine you are constantly losing and regaining signal. You don't want to constantly dismiss that annoying pop up! There are already other visual and audible cues. No need for the pop up!

https://discuss.ardupilot.org/t/constant-connection-lost-connection-restored-pop-ups-on-android-bluetooth/74306

jafrado, 2021-08-24:
It used to just annunciate … I guess your point is that removing all those modal windows is a real pain in the ass while flying 😀

Normally, that should only ever really be an issue if you’re facing radio problems or the FC firmware is right on the edge of the 1 second heartbeat time out (e.g. keep alive packet delayed because Firmware is overloaded). I’ve seen this a few times with balloons over 100,000 feet (on the edge of the fresnel zone) and in other cases it generally indicated a radio or antenna problem.

Can you modify your firmware to send keepalives every 950 ms or are you flying right at the edge and just getting a deluge of pop-ups?

DavidMCampbell, 2021-08-24:
There are many scenarios where there could be periods of time (greater than
1 second) where there are no mavlink messages. For example, you have a
waypoint mission that goes around a mountain or hill, or have a few
waypoints that will be blocked by a land mass, or flying right on the edge,
like you said.

Yes, it's annoying to have to click on the modals/popups while flying.
Usually it has announced that it has lost in regained connection multiple
times and I have to rapidly tap ok to clear everything out so I can see the
map again.