mavlink/qgroundcontrol #9845: Edit position of survey

saandycc, 2021-08-30:
Dear team,
Editing the position of a survey, setting a new position to it seems to have no effect (unlike for waypoints).
May you have a look at that?
Thanks in advance!

booo, 2021-09-01:
We are talking about the Edit Position option that works on waypoints but not on complex patterns:

[image]

julled, 2021-09-08:
I am also in need for getting this bug fixed.

julled, 2021-09-12:
i tried to fix this in https://github.com/mavlink/qgroundcontrol/pull/9865

booo, 2021-11-08:
Even tough there was no proper commit body this was fixed by #9865 therefore I'm closing this issue.

closed 2021-11-08 by booo