mavlink/qgroundcontrol #9846: Xcode Build Guide

soufi, 2021-08-30:
Hello,

I'm trying to build the project on Xcode on Big Sur and Mojave and just couldn't. In fact, I am able to generate the xcode project use the `make xcode` command, which is working great. 
Only problem is that I can't build the project on xcode. the build targets were generated automatically (by selecting, Automatically create schemes when prompted when imported the project) by xcode, and I'm using `ALL_BUILD` target. I'm not even sure if that's how I should be doing, I'm really just following my instinct here.... and here's some of the problems i'm facing right now :

On Big Sur I have the following : 
```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/mmintrin.h:33:5: Use of undeclared identifier '__builtin_ia32_emms'; did you mean '__builtin_isless'?

/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/mmintrin.h:33:25: Too few arguments to function call, expected 2, have 0

/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/mmintrin.h:50:19: Use of undeclared identifier '__builtin_ia32_vec_init_v2si'

/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/mmintrin.h:67:12: Use of undeclared identifier '__builtin_ia32_vec_ext_v2si'

...
```

And on Mojave here's what I have: 
```
Apple Mach-O Linker (ld) Error Group

: "MockConfiguration::staticMetaObject", referenced from:
: "MockConfiguration::MockConfiguration(MockConfiguration*)", referenced from:
: "MockConfiguration::MockConfiguration(QString const&)", referenced from:
: "MockConfiguration::MockConfiguration(QString const&)", referenced from:
: "MockLink::staticMetaObject", referenced from:
: "MockLink::startGenericMockLink(bool, MockConfiguration::FailureMode_t)", referenced from:
: "MockLink::startAPMArduSubMockLink(bool, MockConfiguration::FailureMode_t)", referenced from:
: "MockLink::startAPMArduSubMockLink(bool, MockConfiguration::FailureMode_t)", referenced from:
....
: "MockLink::~MockLink()", referenced from:

: Linker command failed with exit code 1 (use -v to see invocation)

```

I couldn't find any guide for xcode builds, It's been weeks that I'm trying to investigate this but I can't seem to find any solution, so I hope to find some support here. 

Thank you in advance.

louisrm, 2021-08-31:
Hi! Here are two resources I found helpful:
https://dev.qgroundcontrol.com/master/en/getting_started/
https://doc.qt.io/qt-5/ios.html

This is the basic process:
- Install Qt 5.12.6 and include iOS/macOS and Qt Charts in your installation
- Install gstreamer if you want it
- Clone QGC and updating submodules 
- Running qmake to build the .xcodeproj (for example: ``` ~/Qt/5.12.6/iOS/bin/qmake  /path/to/qgroundcontrol.pro CONFIG+=WarningsAsErrorsOn```)
- Open .xcodeproj in Xcode
- Enter developer credentials
- Build with Xcode

soufi, 2021-09-02:
Hi @louisrm ! 
Thank you so much for your prompt response, and apologies for not getting back to you sooner. 

I'm very new to Qt, so apologies in advance if I make some wtf/rtfm mistakes. 

I followed your instructions, I can already tell that I wasn't using the `qmake` from Qt. 
I couldn't find Qt v5.12.6 available on Qt servers so I had to compile it myself. The generated folder didn't contain any iOS folder. 
Instead, I used the following in shadow compile mode: 
```
/usr/local/Qt-5.12.6/bin/qmake ../qgroundcontrol.pro CONFIG+=WarningsAdErrorsOn
```
and it generated the following files in my build folder :
```
Makefile                 qmake_qmake_qm_files.qrc
```
Then I executed a simple `make` command in the build project which triggers a compilation of the project. I have a `qgroundcontrol.app` file in build folder, but no xcode project was generated. 

I'm not sure how to get the iOS folder inside my Qt folder ? Is that the main reason i'm not having the outcome I should ? can't find anything about it in the qt-ios [documentation](https://doc.qt.io/qt-5/ios.html).

Best,

soufi, 2021-09-02:
I used the following command, and it's generating the `. xcodeproj ` file now : 
```
qmake -spec macx-xcode ../qgroundcontrol.pro CONFIG+=WarningsAsErrorsOn
```

closed 2025-05-03 by HTRamsey