mavlink/qgroundcontrol #9847: Cannot run v4.1.4 AppImage: GLIBCXX_3.4.26 and GLIBC_2.29 not found

mortenfyhn, 2021-08-31:
I cannot run the uploaded v4.1.4 AppImage.

I'm on Ubuntu 18.04.

* Download v4.1.4 AppImage from the github release page [here](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.1.4). (The one from the user guide [here](https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html#ubuntu) is currently identical (md5sum), so this applies to both.)
* `chmod +x ./QGroundControl.AppImage`
* `./QGroundControl.AppImage`

gives

```
/tmp/.mount_QGroundHDUWS/QGroundControl: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /tmp/.mount_QGroundHDUWS/QGroundControl)
/tmp/.mount_QGroundHDUWS/QGroundControl: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_QGroundHDUWS/QGroundControl)
```

The current daily build (appears to be 048081c) runs fine. Maybe v4.1.4 is build with a different compiler / otherwise different version of something, which makes it incompatible with Ubuntu 18.04.

I also recall v4.1.3 running fine, but I cannot find a downloadable AppImage for that version anymore. Is it possible to build one and attach it to the github release for v4.1.3?

I observe this issue on multiple different computers running Ubuntu 18.04.

Appears similar to [this](https://docs.qgroundcontrol.com/master/en/Support/troubleshooting_qgc.html#glibc_2_27) problem for Ubuntu 16.04 users.

Cristian-wp, 2021-09-08:
Same problem...do you find any solution?

mehmetyldz87, 2021-09-13:
Hi 
We have same problems. 

https://discuss.px4.io/t/latest-qgc-release/23972/4

patrickelectric, 2021-09-13:
Can you try https://github.com/mavlink/qgroundcontrol/suites/3677341316/artifacts/89532134 ?
It's the artifact from #9853

mortenfyhn, 2021-09-13:
> Can you try https://github.com/mavlink/qgroundcontrol/suites/3677341316/artifacts/89532134 ?
> It's the artifact from #9853

Runs properly on my machine!

mehmetyldz87, 2021-09-14:
> Can you try https://github.com/mavlink/qgroundcontrol/suites/3677341316/artifacts/89532134 ?
> It's the artifact from #9853

It works properly . Thanks @patrickelectric

patrickelectric, 2021-09-14:
Fixed by #9853

closed 2021-09-14 by patrickelectric

dkati, 2021-10-15:
Sorry for reopening this one but, downloading QGC from https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html#ubuntu
produces the safe problem

adfernandes, 2021-11-17:
The problem still exists on both the AWS and Github releases.

Kametor, 2024-10-03:
I am currently working with Ubuntu 22.04 and encountering an issue when attempting to run the QGroundControl Daily build using the following command:

shell
Kodu kopyala
~$ ./QGroundControl-x86_64.AppImage
The following error message is displayed:

bash
Kodu kopyala
./QGroundControl-x86_64.AppImage: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by /tmp/.mount_QGrounD8Qsvv/usr/bin/../lib/libexiv2.so.28)
Based on my understanding, this appears to be related to compatibility issues with the version of the GNU Standard C++ Library (libstdc++.so.6). It seems that the required version, GLIBCXX_3.4.32, is missing or outdated in my current environment.

Could you advise on the best approach to resolve this issue? I would appreciate guidance on either updating the required libraries or any other workaround that may help.

Thank you in advance for your assistance.

Kametor, 2024-10-03:
I solved the problem updating the library
https://stackoverflow.com/questions/76974555/glibcxx-3-4-32-not-found-error-at-runtime-gcc-13-2-0