mavlink/qgroundcontrol #9856: Run on arm64

SEUZTh, 2021-09-05:
I am using ubuntu20.04 arm64 in parallels desktop 17.0.
However, QGC.appimage can't run on arm64.

Do you intend to provide arm64 version?

booo, 2021-09-27:
Can you change the topic to "Support for arm64" platform?

SEUZTh, 2021-09-30:
> Can you change the topic to "Support for arm64" platform?

Ok, I have changed it.

SEUZTh, 2021-10-11:
> Did you try building QGC for arm64?
> 
> https://dev.qgroundcontrol.com/master/en/getting_started/

I am using arm64 ubuntu. If you have a x64 PC, could you x64-arm64 cross build it? Thanks very much.

mfry90, 2022-02-26:
@booo Ditto on the request.  The instructions for building QGroundControl require Qt5, which is also not pre-compiled for arm64. I've tried numerous times on Nvidia Jetson to build qt5, but the build fails, and I'm not sure why (perhaps memory issues?). It would be a huge help if a compiled version for arm64 were already provided.

mengchaoheng, 2022-03-30:
@dagar

nihavend, 2022-04-11:
I am able to run QGroundControl on macos and the px4 on parallels ubuntu 20.04 arm box successfully.

mengchaoheng, 2022-04-11:
> I am able to run QGroundControl on macos and the px4 on parallels ubuntu 20.04 arm box successfully.

How about QGC on arm-based Ubuntu？ I know px4 sitl can be builded on it, and qgc also can run on Mac by Rosetta.

mengchaoheng, 2022-04-18:
@SEUZTh @mfry90 @hyperion11 and @ everyone who need arm- based qgc:
 [summary as](https://thopiekar.eu/uav/qgroundcontrol-on-flatpak/):
1.  sudo apt install flatpak
2. flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo
3. flatpak install flathub org.kde.Platform/aarch64/5.15
4. flatpak remote-add --if-not-exists thopiekar.eu https://dl.thopiekar.eu/flatpak/_.flatpakrepo
5. flatpak install thopiekar.eu org.mavlink.qgroundcontrol

mengchaoheng, 2022-04-18:
> @SEUZTh @mfry90 @hyperion11 and @ everyone who need to run qgc on arm-based ubuntu: [summary as](https://thopiekar.eu/uav/qgroundcontrol-on-flatpak/):
> 
> 1. sudo apt install flatpak
> 2. flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo
> 3. flatpak install flathub org.kde.Platform/aarch64/5.15
> 4. flatpak remote-add --if-not-exists thopiekar.eu https://dl.thopiekar.eu/flatpak/_.flatpakrepo
> 5. flatpak install thopiekar.eu org.mavlink.qgroundcontrol
>    after that, maybe restart your computer is need.

nihavend, 2022-04-18:
> > I am able to run QGroundControl on macos and the px4 on parallels ubuntu 20.04 arm box successfully.
> 
> How about QGC on arm-based Ubuntu？ I know px4 sitl can be builded on it, and qgc also can run on Mac by Rosetta.

unfortunately no. I checked your latest post followed the steps completely without error. But can not start qgc, what is the command ?

mengchaoheng, 2022-04-18:
> 5\. after that, maybe restart your computer is need.

after [that](https://github.com/mavlink/qgroundcontrol/issues/9856#issuecomment-1101268462), maybe restart your computer is need.

mengchaoheng, 2022-04-18:
> after [that](https://github.com/mavlink/qgroundcontrol/issues/9856#issuecomment-1101268462), maybe restart your computer is need.

> unfortunately no. I checked your latest post followed the steps completely without error. But can not start qgc, what is the command ?

after [that](https://github.com/mavlink/qgroundcontrol/issues/9856#issuecomment-1101268462), maybe restart your computer is need.

mengchaoheng, 2022-04-18:
@nihavend

nihavend, 2022-04-18:
> @nihavend

I restarted my parallels vm. But no command for qgc

mengchaoheng, 2022-04-18:
> > @nihavend
> 
> 
> 
> I restarted my parallels vm. But no command for qgc

is there anything output on terminal when you do that 5 step?

nihavend, 2022-04-18:
> mengchaoheng

Because I had a restart I lost the output but executed one ore time : 

`
parallels@ubuntu-linux-20-04-desktop:~/dev/github$ flatpak install thopiekar.eu org.mavlink.qgroundcontrol
Looking for matches…
Skipping: org.mavlink.qgroundcontrol/aarch64/stable is already installed
`

mengchaoheng, 2022-04-18:
> > mengchaoheng
> 
> 
> 
> Because I had a restart I lost the output but executed one ore time : 
> 
> 
> 
> `
> 
> parallels@ubuntu-linux-20-04-desktop:~/dev/github$ flatpak install thopiekar.eu org.mavlink.qgroundcontrol
> 
> Looking for matches…
> 
> Skipping: org.mavlink.qgroundcontrol/aarch64/stable is already installed
> 
> `

Find qgc in lower left quarter of Ubuntu，just klick it

nihavend, 2022-04-18:
> Find qgc in lower left quarter of Ubuntu，just klick it

I thought it was set to path as executable or something similar. Never thought it will be installed as an app. Thanks @mengchaoheng.

I was just leaving working on M1. Do you have a slack account ?

mengchaoheng, 2022-04-21:
> I thought it was set to path as executable or something similar. Never thought it will be installed as an app. Thanks @mengchaoheng.
>
> I was just leaving working on M1. Do you have a slack account ?
Run qgc by 'flatpak run --device=all org.mavlink.qgroundcontrol', see 

> https://github.com/mavlink/qgroundcontrol/issues/10205#issuecomment-1104257576

nihavend, 2022-04-25:
> flatpak run --device=all org.mavlink.qgroundcontrol

I made it work as you mentioned before via app search and tried your last advice from command line. Thanks @mengchaoheng

singulardroid, 2022-09-16:
I'm trying to install on ubuntu 20.04 arm64 and have following:
[image]

singulardroid, 2022-09-16:
looks like the QCG app has dependency on specific version of KDE and more recent versions are not being accepted? too fresh for the subject but honestly can't make it run...

mengchaoheng, 2022-10-23:
> looks like the QCG app has dependency on specific version of KDE and more recent versions are not being accepted? too fresh for the subject but honestly can't make it run...

@singulardroid  Now this method doesn't work because the author seems to have removed this repository

Abdelnacerr, 2022-11-12:
> @SEUZTh @mfry90 @hyperion11 and @ everyone who need to run qgc on arm-based ubuntu: [summary as](https://thopiekar.eu/uav/qgroundcontrol-on-flatpak/):
>
> 1. sudo apt install flatpak
> 2. flatpak [...]

If using mac M1 and have ubuntu installed on parallels, this works like a charm.

mengchaoheng, 2022-11-12:
> If using mac M1 and have ubuntu installed on parallels, this works like a charm.

is work now? I have been test a few days ago but failed since the author remove it.

Abdelnacerr, 2022-11-12:
> is work now? I have been test a few days ago but failed since the author remove it.

yes mate, worked for me with no problems, literally minutes ago.

mengchaoheng, 2022-11-16:
```
parallels@ubuntu-linux-20-04-desktop:~$ flatpak install flathub org.kde.Platform/aarch64/5.15
Looking for matches…
error: Nothing matches org.kde.Platform in remote flathub
parallels@ubuntu-linux-20-04-desktop:~$ 
```
@Abdelnacerr can you help me?

Abdelnacerr, 2022-11-17:
> I can't install org.kde.Platform/aarch64/5.15, how did you do that?
> @Abdelnacerr <https://github.com/Abdelnacerr>

Hey, I installed parallels on desktop,  then I installed ubuntu, then I
cloned the ardupilot repo, then followed the build instructions on their
website.
I am free on saturday from 2pm to 6pm australia time if you're and we can
have a look at it then.

Cheers.

mengchaoheng, 2022-11-17:
> Hey, I installed parallels on desktop, then I installed ubuntu, then I cloned the ardupilot repo, then followed the build instructions on their website. I am free on saturday from 2pm to 6pm australia time if you're and we can have a look at it then. Cheers.
> […](#)

I mean how to install QGC

mengchaoheng, 2022-11-17:
> Hey, I installed parallels on desktop, then I installed ubuntu, then I cloned the ardupilot repo, then followed the build instructions on their website. I am free on saturday from 2pm to 6pm australia time if you're and we can have a look at it then. Cheers.
> […](#)

it work now .juet install flatpak by :
```       sudo add-apt-repository ppa:flatpak/stable
       sudo apt update
       sudo apt install flatpak
```
1.12.4 or later is need

mengchaoheng, 2022-11-17:
> looks like the QCG app has dependency on specific version of KDE and more recent versions are not being accepted? too fresh for the subject but honestly can't make it run...
install flatpak by
```
       sudo add-apt-repository ppa:flatpak/stable
       sudo apt update
       sudo apt install flatpak
```
the 1.12.4 version or later is need

mengchaoheng, 2022-11-17:
> @singulardroid  and @ everyone who need to run qgc on arm-based ubuntu: [summary as](https://thopiekar.eu/uav/qgroundcontrol-on-flatpak/):
> 

1. Must be install flatpak 1.12.4 or later. To install 1.12.4, run the following in a terminal:
```       
sudo add-apt-repository ppa:alexlarsson/flatpak
sudo apt update
sudo apt install flatpak
```
2. `flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo`
3. `flatpak install flathub org.kde.Platform/aarch64/5.15-21.08`
4. `flatpak remote-add --if-not-exists thopiekar.eu https://dl.thopiekar.eu/flatpak/_.flatpakrepo`
5. `flatpak install thopiekar.eu org.mavlink.qgroundcontrol`
   after that, maybe restart your computer is need.

Jcwscience, 2023-03-19:
@mengchaoheng I get the error invalid compressed data. Is this flatpak still being maintained?

mengchaoheng, 2023-03-19:
> @mengchaoheng I get the error invalid compressed data. Is this flatpak still being maintained?

I don't know, sorry about that.

dongdongbh, 2023-05-17:
I just installed the flatpak version, but it can not connect to my px4

elenaponce, 2023-06-05:
> @mengchaoheng I get the error invalid compressed data. Is this flatpak still being maintained?

Hi, where you able to fix this?

sidharthmohannair, 2024-03-27:
# Installing QGroundControl on Ubuntu ARM64 using Flatpak

This tutorial will guide you through the process of installing QGroundControl (QGC) on Ubuntu ARM64 using Flatpak. QGroundControl is an open-source ground control station for small unmanned vehicles that allows you to plan and execute autonomous missions.

https://github.com/sidharthmohannair/Installing-QGroundControl-on-Ubuntu-ARM64-using-Flatpak.git

mengchaoheng, 2024-04-15:
@sidharthmohannair If I want to install a specific version, such as QGC4.1, how do I install it?

sidharthmohannair, 2024-04-15:
> @sidharthmohannair If I want to install a specific version, such as QGC4.1, how do I install it?

Hi there,

I think QGroundControl 4.1 is not available in the standard Flatpak repository. 

Check Available Versions: Use the Flatpak command-line interface to check if the specific version you want is available:

`flatpak search qgroundcontrol`

However, you have a few options to install it:

**Package Manager:** Check if QGroundControl 4.1 is available through your Linux distribution's package manager. Some distributions may have QGroundControl packaged and available for installation through their official repositories.

**Compile from Source:** If neither of the above options is available, you can download the source code of QGroundControl 4.1 from the project's GitHub repository and compile it yourself. This process may require some familiarity with building software from source code and might have additional dependencies.

**Flatpak Custom Repository:** If you are experienced with creating Flatpak packages, you could create a custom Flatpak package for QGroundControl 4.1 and host it in your own Flatpak repository. This would allow you to install it using Flatpak on your system. However, this option requires some expertise in creating Flatpak packages and maintaining a repository.

Choose the option that best fits your preferences.

mengchaoheng, 2024-04-15:
A new method to fill this gap, if you use m1 mac, and want to run sitl on Ubuntu VM, you can run the step:
1. Make sure your vm and mac in the same local network, for example, setup the network in parallels desktop.

2. Run following instructions on Vm
```Console
make px4_sitl gazebo
```
3. After sitl is running, run
```Console
mavlink stop-all

mavlink start -o 14550 -t [your Mac ip] -r 100000 #or other param instead of 100000
```
And now open a qgc on Mac, you can use it.

closed 2025-05-03 by HTRamsey