mavlink/qgroundcontrol #9858: MAVLink Console applies commands to all connected drones (dangerous)

ArkadiuszNiemiec, 2021-09-07:
## Expected Behavior
I expect the QGC to send commands to the currently selected drone.

## Current Behavior
Command is sent to all connected drones, I don't know what reply does it display.
Example: I had 2 connected quadcopters, sent `pwm status`, got one response, tried `pwm test -a -p 1200` and both drones started to spin their motors. 

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Connect multiple drones to QGC
2. Open MAVLink Console
3. Send any commands: 'reboot' is safe example

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: Daily from few weeks ago
- QGC build: daily
- Flight Controller: Cube Orange
- Autopilot (with version): PX4 1.12.0

MAD-CRAZY-MAN, 2021-09-08:
check system id

ArkadiuszNiemiec, 2021-09-08:
> check system id

All my drones have different `MAV_SYS_ID`.

WTPENGUIN, 2021-09-09:
Could you please tell me how to reproduce it in detail?

I configure Cube Black and Pixhawk4 with different Mavlink System IDs and connected to PC with USB.

The Mavlink Console did the right thing, issuing commands only for the selected vehicle.

ArkadiuszNiemiec, 2021-09-21:
> Could you please tell me how to reproduce it in detail?
> 
> I configure Cube Black and Pixhawk4 with different Mavlink System IDs and connected to PC with USB.
> 
> The Mavlink Console did the right thing, issuing commands only for the selected vehicle.

I have two Orange Cubes connected via serial port to RPi4 and proxied by mavros to PC by UDP. Here's the simple test I did (turn on the sound):
https://youtu.be/HRAeL-TEROA
I have also tried connecting them directly using USB cable, same happened.