mavlink/qgroundcontrol #9869: Video streaming can't be changed when connected to a vehicle

hamishwillee, 2021-09-15:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Please describe what happens instead of the expected behavior.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.
2.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: [e.g. 3.4.4]
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

booo, 2021-09-27:
Please provided detailed information about the bug.

zdanek, 2022-11-17:
@booo I've checked and this problem no longer exists. To be closed.

booo, 2022-11-17:
Thanks for checking.

closed 2022-11-17 by booo