mavlink/qgroundcontrol #9870: Automatic selection of video stream broken

hamishwillee, 2021-09-15:
QGC should automatically set up and display "a" video stream for a vehicle that supports the MAVLink Camera API. It appears to correctly select the correct video source but does not display it.

Tested using the [PX4 Gazebo Typhoon simulation](https://docs.px4.io/master/en/simulation/gazebo.html#video-streaming) running on Ubuntu 20.04 (in a VM) against the current release QGC.

## Expected Behavior

When QGC connects to a vehicle that supports the MAVLink Camera API it can tell what video streams it supports. These should be used to configure the application video settings such that the video streams appears in the fly view video widget (irrespective of current video settings).

Because you have auto-configured video streams, the video settings in **Application Settings > General Settings > Fly View** should not be displayed when you are connected to a vehicle that supplies this information.

> **Note:** It isn't clear
>   - How you would turn off video if it was supplied from vehicle but you did not want it.
>  - How it is expected to handle multiple video streams. 
   
## Current Behavior

1. When connected to a vehicle that supports the Camera API, the application video settings to specify the source are removed. This is correct, because you should be getting the settings from the vehicle.
2. However if the application video settings are "wrong" before you connect, then QGC does  _not_ not switch tot he current settings.
   - If the pre-connection setting is "no video", then the video widget is not enabled (video does not stream)
   - If the pre-connection setting is "the wrong video":
     - the video does not stream, but displays the number 1.3 in the video switcher widget
     - When you disconnect and go back to the video settings you can see that the selected video settings is what you would like it to have switched to: *UDP H.264* . 
3. If you try connect again (or if the original setting was correct) then the vehicle WILL stream.

So QGC is switching to the right setting, but doesn't seem to be able to then start viewing the stream.

The fact that this is broken is particularly bad, because it means that if your initial setting is wrong, you can't even go back and fix it.

## Steps to Reproduce:

Video streaming set up exactly as described in PX4 user guide: https://docs.px4.io/master/en/simulation/gazebo.html#video-streaming
PX4 using latest version of master. 
All on Ubuntu 20.04

Only tested for this scenario - so maybe it would switch correctly for some other system

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04 in VMWare VM on Windows 10
- QGC Version: 4.? The current proper release - the general tab  contains incorrect information so can't be precise.
- QGC build: stable.
- Flight Controller: Gazebo Typhoon Simulation
- Autopilot (with version): PX4 master (last stable is v1.12)

## Log Files and Screenshots

Here I set QGC video source to Parrot Discovery (incorrect)
[image]

This is the screenshot of QGC video widget when when I start simulation. Streaming is happening, but not Parrot format:

[image]
Full screen
[image]

Here I disconnect the vehicle, and go back to the settings. Note, they have changed to be correct!
[image]

If I run again, then the video displays correctly

richtong, 2021-09-16:
Thanks @hamishwillee for this explanation. So I'm a little confused by the workaround for this. 

Is this something that gets fixed fast or should we fix the documentation to note this?

hamishwillee, 2021-09-16:
Who knows. I think we wait on @DonLakeFlyer to tell us. Though if @julianoes was working on this for some particular job it might come earlier.

julianoes, 2021-09-16:
For me that's fixed with https://github.com/PX4/PX4-SITL_gazebo/pull/801. QGC then crashes after a bit but that's going to be another issue.

closed 2025-05-03 by HTRamsey