mavlink/qgroundcontrol #9877: QGC Governance and Maintenance

DonLakeFlyer, 2021-09-17:
What is needed to remove me as a single point of failure on QGC maintenance and Governance:

* Top level process manager
  * Pull request tracking/maintenance
  * Issue tracking/maintenance
  * Dev meeting ownership
  * Release management
* Customer Support
  * PX4 QGC Forum
  * ArduPilot QGC Forum
  * Slack  
* High level source areas which require a dedicated maintainer
  * Build
  * CI
  * Localization
  * Video
  * Connectivity
  * Firmware upgrade
  * Base architecture
  * Top level views
    * Fly
    * Plan
    * Vehicle Setup
      * PX4
      * ArduPilot
    * App Settings

booo, 2021-09-23:
If you like me to I can start going through the issues and try to clean them up a bit. Maybe I can do it on a bi-weekly basis so people have the chance to give feedback and/or can help in the process.

I would like us to introduce some tags for bugs and features too.

DonLakeFlyer, 2021-09-23:
That would be fantastic. Thanks.

booo, 2021-09-26:
I guess I need some permissions for that. Could you take a look at that.

I can start to add tags to all tickets in the next days.

DonLakeFlyer, 2021-09-27:
@booo I gave you Triage access. Not sure if that is enough. Give it a try and let me know.

booo, 2021-11-09:
Can you grant me permission to trigger workflows on pull requests?

DonLakeFlyer, 2021-11-09:
> Can you grant me permission to trigger workflows on pull requests?

Are you talking about the case where someone submits a pull request for the very first time and CI doesn't automatically kick off?

booo, 2021-11-09:
> > Can you grant me permission to trigger workflows on pull requests?
> 
> Are you talking about the case where someone submits a pull request for the very first time and CI doesn't automatically kick off?

Yes

DonLakeFlyer, 2021-11-09:
Ok. Let me look again then. I can't seem to find the case for that.

DonLakeFlyer, 2021-11-09:
So it seems to take full repo access to be able to do that which doesn't make much sense to me. Instead I bumped the level where this happens to happen less often. With previous CI there was not such thing at all so I think that is fine to not be so worried about it. And thanks again for all the help.

closed 2024-06-04 by HTRamsey