mavlink/qgroundcontrol #9884: Joystick mapping configuration is not handled correctly

PPakalns, 2021-09-24:
## Expected Behavior
Joystick axis, buttons can be correctly configured and recognized by SDL library.

Joystick GameController configuration environment variable `SDL_GAMECONTROLLERCONFIG` must be correctly processed with the highest priority.

## Current Behavior
Mapping from the SDL_GAMECONTROLLERCONFIG are not recognized.

It could be possible that the loading of GameController mappings from file hardcoded in compilation process overrides value set in SDL_GAMECONTROLLERCONFIG when SDL is initialized with SDL_INIT_GAMECONTROLLER flag.

See:
https://github.com/mavlink/qgroundcontrol/blob/master/src/Joystick/JoystickSDL.cc#L99

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Attach joystick, create custom joystick mappings with different axis order than the default one. 
Mappings can be generated with any tool like https://generalarcade.com/gamepadtool/ . 
2. Check that the gamepad GUID is defined in the resource file to be sure that our configured mapping will be overrided.
https://github.com/mavlink/qgroundcontrol/blob/master/src/Joystick/JoystickSDL.cc#L99
3. Set the environment variable SDL_GAMECONTROLLERCONFIG with the mapping.
4.  Launch QGroundControl, go to joystick view, calibration and check that the mapping  axis order is changed, that the new mapping is loaded. (It won't be)

## System Information
When posting bug reports, include the following information
- Operating System: [Ubuntu 18.04]
- QGC Version: 78cf9bbe6
- QGC build: downloaded AppImage

julled, 2023-03-20:
This would be very handy to map input from USB connected RC controls like the Radiomaster TX16S or Taranis X9D. Currently the axis cannot properly used via QGC.

federicociresola, 2025-04-01:
Is there any news about this bug? Because I am trying to connect my Radiomaster TX16S to QGroudControl, I thought the PC, but QGC maps the radio input incorrectly. I have opened an issue about this #12626

PR #12639 (2025-04-08): Joystick: Allow SDL Mapping from Environment
Allows adding a custom joystick mapping or overriding an existing one via env variable
Closes #9884
Closes #12626