mavlink/qgroundcontrol #9898: QGC closes when disconnecting joystick

rizvanovic, 2021-10-03:
## Expected Behavior
When disconnecting the joystick, the application should not close.

## Current Behavior
Application crashes on disconnect.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Boot up QGC
2. Connect Joystick
3. Disconnect joystick (unplug)
4. Crash QGC

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: 4.1.4
- QGC build: Daily
- Flight Controller: CUAV X7
- PX4 1.12.0

## Detailed Description
Joystick seems to be generally broken.

booo, 2021-10-07:
Could you start QGC from the commandline and provide us with the output? And show us the kernel messages, I guess with "dmesg".

rizvanovic, 2022-10-12:
It definitely still happens 👍

zdanek, 2022-10-14:
Ok. Run QGC with joystick logging
```
./QGroundControl --logging:JoystickLog
```
crash and post logs please

zdanek, 2022-10-14:
and OFC see @booo 's comment

HTRamsey, 2024-06-02:
Most likely a crash due to SDL not handling closing of joysticks correctly

closed 2025-05-03 by HTRamsey