mavlink/qgroundcontrol #9909: Anddroid build error - "make: *** No rule to make target 'install'.  Stop."

iamczar, 2021-10-11:
## Expected Behaviour
I would expect to build and produce an .apk file when I press build.

## Current Behaviour
I build the current master(dd418a5160fcfaaaea94cca9df) for configure for android build and it fails to build and I get this error
"make: *** No rule to make target 'install'.  Stop."

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Clone qgroundcontrol master - commit hash - dd418a5160fcfaaaea94cca9df
2. Configure qt and qtcreator for android build 
3. press build

# Systems
1. Ubuntu 20.04
2. Qt 5.15.2

## Log Files and Screens
[image]
hots
[image]

[image]

I have also tried arm64-v8a

Hi @DonLakeFlyer It was mentioned in the slack channel that I raise this issue here.

Thank you so much in advance.

iamczar, 2021-10-18:
Hello,

I am still getting the same issue - 

I have pulled up to this commit - https://github.com/mavlink/qgroundcontrol/commit/36c01f3e6ae34b8d29b89706451dc4be38b45f1e

I feel like its a setting issue?? Im not sure what Im doing wrong?

minjerd-riis, 2021-10-18:
For me it's generating `staging/android-builds/lib` containing empty `arm64-v8a`, `armeabi-v7a`, `x86`, and `x86_64` folders at the same time that the .so files are generated in . The `libQGroundControl_armeabi-v7a.so` file is copied into `staging/android-build/libs/armeabi-v7a/` and the `libQGroundControl_arm64-v8a.so` file is copied into `staging/android-build/libs/arm64-v8a/`.

I'm not sure why it's not creating the folders for you. Maybe try creating them manually for now just to see if you can get the APK to build.

iamczar, 2021-10-19:
I find that the `staging/android-build/libs/ ` get generated automatically when I disable the make install step 

I then copy the ABI binaries and hit the build again and it creates the only debug apk file - There is no release folder :/
[image]

iamczar, 2021-10-19:
for me its both

minjerd-riis, 2021-10-19:
Yes, it happens for both signed and unsigned. I have not been able to get it to build without disabling the step and manually copying the binaries (for clean build), and I've tried dozens of combinations of settings and versions, working at it almost constantly since Thursday trying to see what works and what doesn't. I'm working with a couple others who have the same issue, one on Windows and one on Linux.

DonLakeFlyer, 2021-10-19:
Ok, great. I'll work on fixing that up as I work towards getting app bundles to work today.

new-user-name, 2021-12-05:
```
!AndroidBuild {
    # See QGCPostLinkCommon.pri for details on why MacBuild doesn't use DESTDIR
    DESTDIR = staging
}
```
Doesn't work on Windows 10, if I get everything right.
`make: *** No rule to make target 'install'.  Stop.`

DonLakeFlyer, 2021-12-06:
If this is a Windows only problem it's not going to much of a priority to fix. Things are working fine on Linux based systems like Ubuntu and OSX which is where Android gets built primarily. CI is on Ubuntu. Certainly happy to take a fix if you find one. But building Linux stuff (like Android) on Windows is always the hardest to make work correctly. Much simpler to use a Linux VM and do it more natively.