mavlink/qgroundcontrol #9969: Not getting OBSTACLE_DISTANCE message in QGC from mavros px4

arpitpara, 2021-10-30:
Hello!

I'm trying to do collision prevention using companion computer setup but I'm not getting any OBSTACLE_DISTANCE mavlink message in QGC.

I even tried to send "rosservice call /mavros/cmd/command .." command to send dummy obstacle messages. It is pinging in COmmand_ack tab with result 4 while broadcast: true and result 3 when broadcast: false.
Can somebody explain what I'm missing?

booo, 2021-10-30:
Please provide detailed steps to reproduce the problem. Please provide additional information about your systems too, information such as QGC version, PX4 version etc. Otherwise we can't reproduce the problem and have to close the issue at some point.

arpitpara, 2021-10-30:
QGC Daily Version : v4.14
Firmware version : 1.12.3 Offical stable release
I followed the offical link provided in Px4 Docs to do collision prevention.

https://docs.px4.io/master/en/computer_vision/collision_prevention.html

ROS Noetic and UBUNTU 21 MATE on Raspberry Pi 4B model as Companion Computer.

I've connected to FCU by USB Port.I'm also able to control the QUADCopter through QGC. Also Command_Ack tab in MAVLINK Inspector shows that it is recieving the message. But OBSTACLE_DISTANCE message is not showing up.

closed 2026-10-04 by github-actions[bot]