mavlink/qgroundcontrol #9972: QGroundControl not compatible with MacOS Monterey 12.0.1

SamuelSHernandez, 2021-11-01:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.
When launching QGroundControl with macOS Monterey 12.0.1, the application should open with no issues. 

## Current Behavior
Please describe what happens instead of the expected behavior.
the main map does not load. Instead, there's just a small logo and white screen.
## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Update OS to macOS Monterey 12.0.1 
2. Open QGroundControl

## System Information
When posting bug reports, include the following information
- Operating System: macOS Monterey 12.0.1 
- QGC Version: 4.1.4
- QGC build: stable
- Flight Controller: Pixhawk 4
- Autopilot (with version): PX4

## Detailed Description
Provide further details about your issue/bug.
I updated my computer and could no longer open the app correctly

## Log Files and Screenshots
[image]
[image]

DonLakeFlyer, 2021-11-02:
Try this build and let me know what happens: https://qgroundcontrol.s3.us-west-2.amazonaws.com/DonLakeFlyer/QGroundControl.dmg

cbjornram, 2021-11-04:
I saw this and thought I'd tune in. I'm also running macOS 12.0.1 with QGC 4.1.4 and for me it's working without issues, both with the official download (I tried with a fresh install) and with the build you posted above @DonLakeFlyer. 

@SamuelSHernandez Have you selected a map provider that has coverage in your area? And what hardware are you on? I'm running my tests on MacBook Pro 2015 (Intel).

DonLakeFlyer, 2021-11-04:
Also FYI top folks. Daily build are now posting again for download from the QGC site. So you don't need to use the dmg I linked to. You an just use a regular daily to test with now.

SamuelSHernandez, 2021-11-04:
I think the issue was the map provider. I tried the build posted above and it worked. I will close the issue now.

closed 2021-11-04 by SamuelSHernandez

SilaLiu, 2023-03-29:
mac系统为13.0的目前不能使用QGC

SilaLiu, 2023-03-29:
> 试试这个构建，让我知道会发生什么：https://qgroundcontrol.s3.us-west-2.amazonaws.com/DonLakeFlyer/QGroundControl.dmg

链接出现了错误，下载不了

cbjornram, 2023-03-29:
No that link is not really relevant anymore since there have been more recent versions released since.

What's your issue, @SilaLiu? Is it regarding maps not showing, or something else? Unfortunately I don't have a mac that can run macOS 13 so I can't really test anything at the moment.