mavlink/qgroundcontrol #9975: Wrong spelling of words in QGroundControl document

qiang-sir, 2021-11-02:
## Expected Behavior
Modify the spelling of the word "overries" which is shown in the attached picture.

## Current Behavior
The word "overrides" is spelled "overries".

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.Locate to Top Level Views
2.Locate to Plan Views
3.Find the word position as shown in the picture
4.Modify the spelling of the word "overries"
[image]

cbjornram, 2021-11-04:
I created a PR for you:
https://github.com/mavlink/qgc-dev-guide/pull/138

hamishwillee, 2021-11-04:
Thanks! Create a PR the right thing to do. Much appreciated.

closed 2021-11-04 by hamishwillee