mavlink/qgroundcontrol #9983: Switch default MAVLink dialect to development.xml

bkueng, 2021-11-03:
Since MAVLink switched away from WIP flags, and instead uses a `development.xml` dialect, using that will simplify integration of new features.
@DonLakeFlyer as discussed.

QGC currently still uses a few messages from ardupilotmega in Vehicle (grep for `NO_ARDUPILOT_DIALECT`), and FirmwarePlugin/APM/* among others.

WTPENGUIN, 2021-11-03:
What is the compatibility with Ardupilot if switch to development.xml?

Does it change so that it cannot be connected to the board on which the Ardupilot series firmware is installed?

bkueng, 2021-11-04:
No, it would preserve compatibility, but we'll need to see how to handle those extra messages.

DonLakeFlyer, 2021-11-04:
> QGC currently still uses a few messages from ardupilotmega in Vehicle (grep for `NO_ARDUPILOT_DIALECT`), and FirmwarePlugin/APM/* among others.

`NO_ARDUPILOT_DIALECT` The naming of that define isn't quite correct. That define indicates all code relating to the Ardupilot firmware. Whether it is related to the mavlink protocol or not. From doing code inspection the only ArduPilot mavlilnk dialect specific stuff is here:
* MAVLINK_MSG_ID_CAMERA_FEEDBACK - https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L769. The mavlink common message for this is MAVLINK_MSG_ID_CAMERA_IMAGE_CAPTURED.
* MAV_CMD_FLASH_BOOTLOADER  - https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L2995. I'm not sure if there is a common equivalent to this.

So give these two things it's not currently possible to switch without losing functionality.

I haven't had time to try a recompile with the common dialect. That would show anything else. But I think this is it. So in order

DonLakeFlyer, 2021-11-04:
> Since MAVLink switched away from WIP flags, and instead uses a `development.xml` dialect, using that will simplify integration of new features.

Since a mavlink dialect is a compile time option what is the benefit of a "development" dialect in combination with a "common" dialect. For example if it means that QGC needs to always be compiled with the "development" dialect doesn't that mean that just becomes the standard and common ends up never being used?

DonLakeFlyer, 2021-11-04:
I did some more poking around. There is more to it than what I said above. A recompile will tell all.

bkueng, 2021-11-05:
> Since a mavlink dialect is a compile time option what is the benefit of a "development" dialect in combination with a "common" dialect. For example if it means that QGC needs to always be compiled [...]

The idea is that you can select at build, and for releases you would select `common` and exclude all features from `development`.

closed 2023-08-24 by DonLakeFlyer