mavlink/qgroundcontrol #9988: Add widget to Map (or flight) view

UcefMountacer, 2021-11-05:
Tell us a bit about the feature:

I just need to know which file I should modify to add a widget on the screen (say a pin) in order to show some location sent by the drone on board computer.

For example, the drone will send the latitude, longitude data of an object, and the GUI will show it on the map.

Can someone show me which file is probably the one and even if possible how to do it ?

Thanks

booo, 2021-11-08:
A team mate implemented something that displays icons on the map. Maybe that is of inspiration to you: https://gitlab.com/searchwing/development/groundstation/searchwing-qgroundcontrol/-/tree/develop/7-add-visualisation-of-detections-from-detections-provider

Maybe you can abuse the ADSB vehicle message or the code around that?

If you are looking for a more general feature to show custom objects/overlays on the QGC map please reopen the ticket or create a feature request with a detailed description of the feature you are looking for.

closed 2021-11-08 by booo

UcefMountacer, 2021-11-12:
Thank you @booo

deltatango2, 2023-04-04:
@booo I am pursuing this same feature of placing items on a map based on user provided data.  I looked at the two methods you described. One looks like I could generate the mavlink adsb message filled with the data of my desired map item.  The other is creation of a manager like searchwing did and add map items from that.  I am leaning toward the searchwing approach because I could have more control of the mapitem features and properties.  
Before I go deeper into code dev, I wanted to verify that as of March 2023 there isn't a better method I should try first.  Thanks!