mavlink/qgroundcontrol #10001: QGroundControl crash when updating SI1060 Radio

vincentpoont2, 2021-11-08:
## Expected Behavior
Sik radio FW should upgrade.

## Current Behavior
Sik radio would not upgrade and system would crash.

## System Information
When posting bug reports, include the following information
- Window 10
- QGC Version 4.1.4
- QGC stable build

## Detailed Description
QGroundControl would crash when updating Holybro SI1060 radio. Support for this radio was added here https://github.com/ArduPilot/SiK.

booo, 2021-11-08:
Can you provide us with some logs? Otherwise this is quite hard to debug.

DonLakeFlyer, 2021-11-08:
Also try a daily build and see if that crashes as well

vincentpoont2, 2021-11-12:
I sent a pair to @DonLakeFlyer . I have ask PX4 developers to assist on finding the source of the FW (http://px4-travis.s3.amazonaws.com/SiK/stable/radio~hm_trp.ihx).

@dagar is helping us to look into this. I suspect that it might be from https://github.com/LorenzMeier/SiK and it need a sync with upstream. Support for this SI1060 Radio has been added to https://github.com/ArduPilot/SiK.

[image]
[image]

DonLakeFlyer, 2021-11-21:
So from this report it's unclear if you were seeing a crash. Does it crash after you get the "did not respond to command mode" report?

For me the radio goes into command mode fine. The problem is that the board id returned for this radio is 80. Whereas all other SiK radios are 78. This cause the firmware for it to not be found. Why is this different if this is going to use the same firmware as other SiK radios?

DonLakeFlyer, 2021-11-21:
I tried simulating a failure at the command mode step and I couldn't cause it to crash with that either.

vincentpoont2, 2021-11-22:
So Sorry @DonLakeFlyer, I have updated the title, the "crash" only happened in Mission Planner and I got confused.
This radio uses a different chip, SI1060 vs SI1000 before, so the board_ID changed.

vincentpoont2, 2021-11-22:
The support was added on here. https://github.com/ArduPilot/SiK. I am still trying to figure out where QGC pull this from via @mrpollo and how to get QGC to be able to perform upgrade for this radio.

DonLakeFlyer, 2021-11-22:
> This radio uses a different chip, this new radio uses SI1060, the previous one uses SI1000, so the board_ID changed.

Ok, understood. I can add support for the new board id but I don't think that makes sense until this: http://px4-travis.s3.amazonaws.com/SiK/stable/radio~hm_trp.ihx get updated to support the new radio. Otherwise it's going to flash incorrect firmware. Is that correct?

mrpollo, 2021-11-30:
@DonLakeFlyer here's the link to the new firmware https://px4-travis.s3.amazonaws.com/SiK/stable/radio~hb1060.ihx

DonLakeFlyer, 2021-11-30:
@mrpollo Where was this built from? If this comes from the ArduPilot repo then it is incorrect with respect to PX4. The PX4 sik radio firmware has updates for mavlink 2 support detection which are not in the ArduPilot repo.

vincentpoont2, 2021-12-01:
This was sent to me by Andrew Tridgell and I am assuming it is built from [https://github.com/ArduPilot/SiK](https://github.com/ArduPilot/SiK).

DonLakeFlyer, 2021-12-01:
So that means it is only "sort of" correct. The PX4 version of Sik radio firmware has additional support for mavlink v2 protocol negotiation whereas the ardupilot version does not. So that feature will be lost. Not a show-stopper necessarily. But the PX4 version of the Sik radio firmware should be brought up to date with the ardupilot changes. And then QGC should switch to using that somewhat better version. For now I think we can live without that feature. Unless this can be taken care of quickly by PX4 folks before I push out a new stable.

vincentpoont2, 2021-12-02:
@mrpollo do you know if this is something PX4 can assist with?

vincentpoont2, 2021-12-03:
@DonLakeFlyer May you add it as it is, then we will check & test it via QGC Daily.

DonLakeFlyer, 2021-12-06:
Yup. That's my plan.

DonLakeFlyer, 2021-12-06:
@vincentpoont2 Would be helpful if you could verify things still working on both SI1000 and SI1060 boards.

closed 2021-12-06 by DonLakeFlyer

vincentpoont2, 2021-12-08:
Hi @DonLakeFlyer I just tried it on both SI1000 & SI1060 boards.

I am not sure why there was a couple instances it failed to upgrade, but at the end, I was able to upgrade on both pair of radios and the connection was successful. 

[image]
[image]
[image]

DonLakeFlyer, 2021-12-08:
@vincentpoont2  Thanks for testing. Those Sync problems are fairly common. Radios (not sure why) seem to be flaky with respect to getting them into bootloader mode. If you retry it will tend to work. I've looked at it in the past and never found anything I could do in QGC that would prevent it from happening.

PR #10064 (2021-12-06): Support 1060 based SiK radios
Fix #10001