mavlink/qgroundcontrol #10002: Error on setting language to Portuguese

kressinluiz, 2021-11-08:
## Expected Behavior
QGC Language should change from English to Portuguese when set on Application Settings.

## Current Behavior
Nothing happens.

## Steps to Reproduce:
1. Go to QGC Menu -> Application Settings -> Miscellaneous -> Language combobox
2. Click on Portuguese.

## System Information
- Operating System: ubuntu 20.04
- QGC Version: master branch 08/11/2021
- QGC build: self-built from source
- Flight Controller: not related

## Detailed Description
Qt Locale identifies my system as pt_BR instead of pt_PT and as the translations sources are defined as pt_PT, QGC throws an error on the console because cannot find sources with pt_BR ending.

I managed to fix this by copying, pasting, renaming the pt_PT sources to pt_BR and adding them to the project.
But I don't know the best way to solve this. If we should add a condition on the code, add one more language to the project exactly the same as pt_PT...

## Log Files and Screenshots
[image]

FYI @hamishwillee

booo, 2021-11-08:
As far as I understand the situation we only manage languages in our translation process and use a combination of language and territory to store the translation. But the logic for choosing the right language takes the territory into consideration too. I guess one solution would be to remove the territory from the logic and only use the language part of the QLocale to search for the right language.

kressinluiz, 2021-11-08:
@booo Thank you for your insights. 

Today we are looking for the translation files with QTranslator::load function:
[image]

Looking through [QTranslator::load docs](https://doc.qt.io/qt-5/qtranslator.html#load-1), it's possible to see that this function tries to load several combinations of filenames + prefix + ui language name + suffix according to the parameters:
[image]

So for example for source localization:
- filename = "qgc_source_"
- prefix = ""
- directory = ":/i18n"
- ui language names = "pt", "pt-BR", "pt-Latn-BR"
The function will look for the following:
- qgc_source_pt
- qgc_source_pt_BR
If we had ui language names for portugal as "pt", "pt-PT", the function will look for:
- qgc_source_pt
- qgc_source_pt_PT

So what do you think we store the translation as a combination of just language and let the QTranslator::load function choose the correct translation file according to the combination of language and territory?

I don't know if we have translations that are dependent to region/territory.

booo, 2021-11-08:
As far as I can see there are no translations that are dependent on the territory yet. Please double check here: 

https://github.com/mavlink/qgroundcontrol/tree/master/translations

closed 2025-05-03 by HTRamsey