mavlink/qgroundcontrol #10006: Incomplete Parameter Detection

ES-Alexander, 2021-11-09:
## Expected Behaviour
All parameters detected when QGC connects to vehicle

## Current Behaviour
Some JS button function values not detected IF QGC is open before the vehicle is turned on+connected.
- In particular, ArduSub 4.0.3 with a joystick button set to `servo_3_min_max_momentary` has that joystick button register as `unknown: 85`. More generally, `BTNn_FUNCTION`s 57-87 and 101-107 seem to all be missing. 
- restarting QGC while the vehicle is already on and connected resolves the issue
- disconnecting the vehicle (physically and then with the "Disconnect" button) then reconnecting it also resolves the issue
- issue only seems to appear when the vehicle has been turned off (e.g. disconnecting an on vehicle, and restarting QGC before reconnecting the vehicle doesn't seem to have the same issue) - this point should be verified independently, because Mac Parallels might be doing something weird with my connections to QGC on Windows

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Connect a joystick to computer (I believe this is also fine after step 2 or 3)
2. Open QGC
3. Power up ArduSub vehicle and connect it to computer running QGC
4. Go to `Vehicle Setup/Joystick/Button Assignment` and check the available button function options (after `relay_2_toggle` should be `relay_3_on`, but is actually `custom_1` (which is also valid, just should come later); last value should be `relay_4_momentary` but is actually `custom_6`)

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10, via Parallels on MacOS 11.6, but also on a normal Windows 10 install
- QGC Version: 4.1.4, and 4.1.2; NOT 4.0.5
- QGC build: stable
- Flight Controller: Pixhawk I
- Autopilot (with version): ArduSub 4.0.3

## Detailed Description
Bug found and discussed a bit in [this forum thread](https://discuss.bluerobotics.com/t/gripper-and-pixhawk-strange-effect-now-dead-gripper-and-qgc-status-odd/10965).

## Log Files and Screenshots
- QGC Console Log:
Log includes initial connection, with missing button functions, then I reconnected the vehicle (described in "Current Behaviour") and the button functions were all available (search for `UDPLink` to find the two connection points).
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.
<figure>
  <figcaption>Missing joystick button functions - Unknown:85:</figcaption>
  [image]
</figure>
<figure>
  <figcaption>Joystick button functions as expected:</figcaption>
  [image]
</figure>

ES-Alexander, 2022-04-08:
I've just confirmed that this issue is still present as of QGC 4.1.7, in both Windows 10 and MacOS (which makes sense, since it hasn't been targeted for a fix).

I did find however that while it's present when connecting via [Companion](https://github.com/bluerobotics/companion), it is _not_ present when connecting via [BlueOS](https://github.com/bluerobotics/BlueOS-docker).

Williangalvani, 2022-04-08:
These issues are usually related to metadata files.
I recently updated metadata for Sub 4.1 in Master (QGC 4.2?):
https://github.com/mavlink/qgroundcontrol/blob/master/src/FirmwarePlugin/APM/APMParameterFactMetaData.Sub.4.1.xml#L2439

One other cause for this is that sometimes QGC fails to identify the vehicle and load the proper metadata. This is usually noticeable as some SUB-specific UI fails to load, and QGC loads "flight modes" and other copter stuff instead.

closed 2026-10-04 by github-actions[bot]