mavlink/qgroundcontrol #10011: I need a translation file (.ts) of PX4ParameterFactMetaData.xml.

muramura, 2021-11-10:
Tell us a bit about the feature:
- What problem does it solve?

Config parameter summary, description, and choices are only in English.
You can understand the config parameter in your own language.

- What does it do?

Displays the config parameter in the language of each country.

- What flight stacks must it work with? (All, PX4, ArduPilot)

PX4

- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)

ALL

- Are there other systems that have this feature?

Ability to display translations of JSON parameters and pop-up messages

- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)

PX4ParameterFactMetaData.xml file generation tool

Any additional context you can provide will make the feature easier to evaluate (e.g. mockups, detailed specification, etc.)

NONE

booo, 2021-11-13:
You can help with the translation efforts over here:

https://crowdin.com/project/qgroundcontrol

For some hints about the process please take a look at: https://github.com/mavlink/qgroundcontrol/issues/9962

As far as I understand you, we should implement translation for the parameters stored in PX4ParameterFactMetaData.xml. Is this right?

muramura, 2021-11-14:
I have already done the translation of the source, JSON.

> As far as I understand you, we should implement translation for the parameters stored in PX4ParameterFactMetaData.xml. Is this right?

Yes, it is correct.

closed 2025-07-04 by HTRamsey