mavlink/qgroundcontrol #10016: Ubuntu 18.04 Stable AppImage for v4.1.4 is Outdated and Unusable

3dpinta, 2021-11-12:
Issue #9847 brought up a the problem of Ubuntu 18.04 users not being able to run the latest version (v4.1.4) of QGC. Issue #9853 address #9847 and appears to fix the problem but as the later comments in Issue #9847 show, the latest github releases and the AppImage linked to in the downloads sections still point to the version that will only work on Ubuntu 20.04. If the correct version can be linked, that would be great.

Information requested:
- Ubuntu 18.04 LTS
- v4.1.4 stable (from the downloads page). Have to roll back to 4.0.11 to get a functioning GCS. There are no AppImage files for 4.1.0-4.1.3 stable available to pick up more recent improvements/fixes.
- Cube Orange, Cube Black
- PX4 latest and ArduCopter 4.0.3
- Downloaded latest stable image, followed instructions at https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html, and when starting via command line and from icon, I continue to get the error from Issue #9847

Jaeyoung-Lim, 2021-11-13:
I can confirm that QGroundcontrol is broken on ubuntu 18.04
```
/tmp/.mount_QGrouneKMv3t/QGroundControl: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_QGrouneKMv3t/QGroundControl)
```

mrpollo, 2021-11-30:
@3dpinta @Jaeyoung-Lim we will be releasing a new stable release this week that will be compatible with Ubuntu 18.04 again. We are sorry for the inconvenience.

3dpinta, 2021-12-02:
@mrpollo Can the correct one at least be linked? If someone needs to use an older version for compatibility purposes, it would be helpful to get to it rather than have to go rebuild it themselves.

patrickelectric, 2022-02-17:
[check 4.1.5 release ](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.1.5)

ES-Alexander, 2022-02-20:
> [check 4.1.5 release ](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.1.5)

This is [confirmed to be working](https://discuss.bluerobotics.com/t/video-streaming-is-not-working-on-linux-pc/11523/12). This issue can be closed.

closed 2022-03-07 by mrpollo

ashishmagar600, 2022-07-04:
> > [check 4.1.5 release ](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.1.5)
> 

Why are you packing in tar.gz if we need to recursively clone from other git repos? It throws error dir mavlink not found. Please make note