mavlink/qgroundcontrol #10017: Daily builds for Mac and Linux are not working.

dlech, 2021-11-12:
The links for Linux and Mac on the page https://docs.qgroundcontrol.com/master/en/releases/daily_builds.html are giving errors:

```
<Error>
<Code>AccessDenied</Code>
<Message>Access Denied</Message>
<RequestId>N85XC9QW4ZCJM5KT</RequestId>
<HostId>
0pjmJk08CbOpfqJjxJkBVAh2lDKFd+X4gRQqSXe5yFbAXx/qrEEhscZagZEt0Fja8KMn6Ez/aqU=
</HostId>
</Error>
```

The Windows link is working.

Here are the URLs for reference:
- Windows: https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl-installer.exe
- Mac: https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl.dmg
- Linux: https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl.AppImage

bols-blue, 2021-11-13:
If you want to download the daily build, you can get it from the artifacts in the link below.
https://github.com/mavlink/qgroundcontrol/actions/workflows/linux_release.yml
https://github.com/mavlink/qgroundcontrol/actions/workflows/macos_release.yml
https://github.com/mavlink/qgroundcontrol/actions/workflows/windows_release.yml

S3 upload doesn't seem to work.
#9946

booo, 2021-11-13:
Yes, I guess that is related to the S3 issues. I close this therefore.

closed 2021-11-13 by booo