mavlink/qgroundcontrol #10019: Compass Calibration Fitness parameter setting error

WTPENGUIN, 2021-11-15:
## Expected Behavior
Compass Calibration fitness parameter's range is 4 ~ 32.

## Current Behavior
Compass Calibration fitness parameter is set 100 in compass calibration.

## System Information
When posting bug reports, include the following information
- Operating System : All System
- QGC Version: 4.1.4
- QGC build: Stable
- Flight Controller: All pixhawk series Flight Controllers
- Autopilot (with version): All ArduCopter firmware

## Detailed Description
The fitness parameter of Compass Calibration in ArduCopter ranges from 4 to 32. But when QGC performs Compass Calibration, it sets a value of 100.  Is this a trick that works?

[image]

booo, 2021-11-16:
Please provide us with a diff instead of a screenshot. It's really hard to work with screenshots.

WTPENGUIN, 2021-11-16:
@booo Sorry. Changed to diff instead of screenshot. Please review.

closed 2026-10-04 by github-actions[bot]