mavlink/qgroundcontrol #10020: Bug Found in msgID #126 mavlink protocol OOB read

BOB4Drone, 2021-11-15:
## Expected Behavior
It seems that the SERIAL_CONTROL packet(msgid 126) oob bug can cause a memory leak or corruction.

## Current Behavior
Please describe what happens instead of the expected behavior.

[image]
In a normal mavlink packet, the length of the data field is within 70. So the value of the count field will also be within 70.
However, if the value of the count field is entered larger than 70, oob read occurs.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. built qgroundcontrol as an asan. I referred to [qgc asan build](https://github.com/mavlink/qgroundcontrol/issues/9626) 
2. send heartbeat packet and serial_control packet(msg id 126)
3. asan detect oob read

## Packet Used to Reproduce this packet

heatbeat packet
```
fd090000000101000000c2b4edc297978c9cee18b7
```

SERIAL_CONTROL packet(msg id 126)
```
fd2c00000001017e000007aeb2e97ffccc50ffAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA3bc4
```

## System Information
- Operating System: ubuntu 20.04
- QGC Version: latest 

## Detailed Description

When gcs receive a serial_control packet, they call mavlinkSerialControl.  
At this time QByteArray function is called and the argument of the function is data, count field in SERIAL_CONTROL packet. 

[code link](https://github.com/mavlink/qgroundcontrol/blob/4c86de5f351cea85b850537ac25cd1ea6c7eb45e/src/Vehicle/Vehicle.cc#L764) 

```
    case MAVLINK_MSG_ID_SERIAL_CONTROL:
    {
        mavlink_serial_control_t ser;
        mavlink_msg_serial_control_decode(&message, &ser);
        emit mavlinkSerialControl(ser.device, ser.flags, ser.timeout, ser.baudrate, QByteArray(reinterpret_cast<const char*>(ser.data), ser.count));
    }
```

  
and inside the QByteArray function there is a memcpy function like this.  
the memcpy function argument is SERIAL_CONTROL packet's data, count field.  

[code link](https://github.com/qt/qtbase/blob/b7f20193573d2a5ae7cb765468023d97dfc37420/src/corelib/text/qbytearray.cpp#L1679) 
```
QByteArray::QByteArray(const char *data, qsizetype size)
{
    if (!data) {
        d = DataPointer();
    } else {
        if (size < 0)
            size = qstrlen(data);
        if (!size) {
            d = DataPointer::fromRawData(&_empty, 0);
        } else {
            d = DataPointer(Data::allocate(size), size);
            Q_CHECK_PTR(d.data());
            memcpy(d.data(), data, size);
            d.data()[size] = '\0';
        }
    }
[... 65 lines cut ...]
  0x10003e40a8a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10003e40a8b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==1922==ABORTING

```

bkueng, 2021-11-16:
Fix: https://github.com/mavlink/qgroundcontrol/pull/10022

BOB4Drone, 2021-11-16:
> Fix: #10022
Can you give me a bug tag for our paper?

booo, 2021-11-16:
Nice catch!

Isn't that a problem of the parser? If the count and the actual data size don't match it seems to me like an invalid message or corrupted message that should be discarded by the parser.

BOB4Drone, 2021-11-16:
yes i think we should add discard routine code in parser if it invalid message or corrupted message.

closed 2021-11-24 by BOB4Drone