mavlink/qgroundcontrol #10026: Bugs on Ubuntu 20 with Daily

farhangnaderi, 2021-11-17:
----

I came up to two bugs as below:

## No. 1:
## Expected Behavior
Responding to map type selection in settings menu. 

## Current Behavior
First time opening QGC it crashes after changing the map type. After that the problem resolved.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Open QGC (Fresh download)
2. Select different map type in settings

##
## No. 2:
## Expected Behavior
Normal param transfer after connecting the FCU. 

## Current Behavior
CPU load high error received and all params transfer stopped.  

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Open QGC (Fresh download)
2. Connect your FCU 
3. Wait for Param transfer

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: Daily
- QGC build: daily
- Flight Controller: Pixhawk 4
- Autopilot (with version): PX4 1.12.3 and master

## Detailed Description

## Log Files and Screenshots
[image]
[image]
[image]

farhangnaderi, 2021-11-19:
tried in another system, same CPU load high.

booo, 2021-11-19:
Please use one issue to describe exactly one problem.

farhangnaderi, 2021-11-20:
Thanks @booo , Corrected.

dagar, 2021-12-14:
Which board is this and what's it running exactly?

farhangnaderi, 2021-12-14:
v5x and Pixhawk4 both same. Running master and also tried with the stable.

dagar, 2021-12-14:
> v5x and Pixhawk4 both same. Running master and also tried with the stable.

Can you get to a console and run `top`? Which airframe?

farhangnaderi, 2021-12-14:
Hardware not at hand now. S500 Frame.

mrpollo, 2021-12-14:
Tested on a VM under macOS using FMUv5X with PX4 master + QGC Daily and I couldn't reproduce, anything else you can share will help @farhangnaderi

farhangnaderi, 2021-12-16:
I changed my Ubuntu (Fresh new one) and everything seems ok with Pixhawk4 and the latest QGC daily. Will open another issue if the problem reproduced.

closed 2021-12-16 by farhangnaderi