mavlink/qgroundcontrol #10028: Using JDK 8 results in build failures

slade-auterion, 2021-11-18:
## Expected Behavior
QGC should build when an appropriate JDK is selected. As per Qt Android 5.15 specifications, _JDK 11 or earlier must be used when using Qt for Android._ . 

## Current Behavior
The following JDK 8 variants have caused Android build step errors, while using Qt 5.15.2 to build QGC
- All variants JDK 8 of Oracles official JDK releases listed here https://www.oracle.com/java/technologies/downloads/#java8 
- All variants of AdoptOpenJDK JDK 8 listed here https://adoptopenjdk.net/ 
- Qt Android toolchains within Windows, Linux or MacOS are all affected.

## Steps to Reproduce:
1. In Qt Creator > Projects > Manage Kits > Devices > Android > Java Settings > JDK location > note the directory of the JDK
2. Check if your systems local JDK installation is at least version JDK 11, if not, its probably JDK 8
3. Run Qt 5.15.2 Android Debug or Release builds for QGC
4. Observe error log which is similar to the log posted here.
5. Reproducable in Windows, Linux or MacOS Qt Android toolchains.

## Steps to Fix
1. Install JDK 11 (any variant, any OS)
2. In Qt Creator > Projects > Manage Kits > Devices > Android > Java Settings > JDK location > point to the local installation folder of JDK 11
3. Apply > OK, and then build Qt 5.15.2 in either Debug or Release mode
4. Error disappears once JDK8 is removed.

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10, MacOS Monterey, Ubuntu 20.04
- QGC Version: develop as of 2021-NOV-18

## Log Files and Screenshots
https://imgur.com/5UkwxbX
https://imgur.com/M1IA0ph

full error log from Windows Android toolchain as follows : 
```

> Task :mergeDebugShaders
> Task :compileDebugShaders
> Task :generateDebugAssets
> Task :mergeDebugAssets
> Task :processDebugJavaRes NO-SOURCE
> Task :checkDebugDuplicateClasses
> Task :mergeDebugJniLibFolders
> Task :validateSigningDebug
> Task :mergeDebugJavaResource
> Task :javaPreCompileDebug
> Task :mergeDebugResources
> Task :processDebugResources

> Task :compileDebugJavaWithJavac FAILED
An exception has occurred in the compiler (1.8.0_311). Please file a bug against the Java compiler via the Java bug reporting page (http://bugreport.java.com) after checking the Bug Database (http://bugs.java.com) for duplicates. Include your program and the following diagnostic in your report. Thank you.
java.lang.AssertionError: annotationType(): unrecognized Attribute name MODULE (class com.sun.tools.javac.util.UnsharedNameTable$NameImpl)
	at com.sun.tools.javac.util.Assert.error(Assert.java:133)
	at com.sun.tools.javac.code.TypeAnnotations.annotationType(TypeAnnotations.java:231)
	at com.sun.tools.javac.code.TypeAnnotations$TypeAnnotationPositions.separateAnnotationsKinds(TypeAnnotations.java:294)
	at com.sun.tools.javac.code.TypeAnnotations$TypeAnnotationPositions.visitMethodDef(TypeAnnotations.java:1066)
	at com.sun.tools.javac.tree.JCTree$JCMethodDecl.accept(JCTree.java:778)
	at com.sun.tools.javac.tree.TreeScanner.scan(TreeScanner.java:49)
	at com.sun.tools.javac.code.TypeAnnotations$TypeAnnotationPositions.scan(TypeAnnotations.java:275)
	at com.sun.tools.javac.tree.TreeScanner.scan(TreeScanner.java:57)
	at com.sun.tools.javac.code.TypeAnnotations$TypeAnnotationPositions.visitClassDef(TypeAnnotations.java:1042)
	at com.sun.tools.javac.tree.JCTree$JCClassDecl.accept(JCTree.java:693)
	at com.sun.tools.javac.tree.TreeScanner.scan(TreeScanner.java:49)
	at com.sun.tools.javac.code.TypeAnnotations$TypeAnnotationPositions.scan(TypeAnnotations.java:275)
	at com.sun.tools.javac.code.TypeAnnotations$1.run(TypeAnnotations.java:127)
	at com.sun.tools.javac.comp.Annotate.flush(Annotate.java:152)
	at com.sun.tools.javac.comp.Annotate.enterDone(Annotate.java:129)
	at com.sun.tools.javac.comp.Enter.complete(Enter.java:512)
	at com.sun.tools.javac.comp.Enter.main(Enter.java:471)
	at com.sun.tools.javac.main.JavaCompiler.enterTrees(JavaCompiler.java:982)
	at com.sun.tools.javac.main.JavaCompiler.compile(JavaCompiler.java:857)
	at com.sun.tools.javac.main.Main.compile(Main.java:523)
[... 123 lines cut ...]
	at java.lang.Thread.run(Thread.java:748)

> Task :desugarDebugFileDependencies
> Task :mergeDebugNativeLibs
> Task :mergeLibDexDebug
22 actionable tasks: 22 executed

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 11s
Building the android package failed!
  -- For more information, run this command with --verbose.
16:20:04: The process "C:\Qt\5.15.2\android\bin\androiddeployqt.exe" exited with code 14.
Error while building/deploying project qgroundcontrol (kit: Android Qt 5.15.2 Clang Multi-Abi)
When executing step "Build Android APK"
```

DonLakeFlyer, 2021-11-21:
Sure would be nice if someone from Auterion helped in figuring out how to fix this.

slade-auterion, 2021-11-21:
> Sure would be nice if someone from Auterion helped in figuring out how to fix this.

is JDK 8 a hard requirement for QGC? Curious if we can make JDK 11 part of the requirements, since these toolchain errors disappear when you upgrade to JDK 11.

DonLakeFlyer, 2021-11-21:
That all has to do with Qt. QGC itself has no requirements for JDK. For Android it's all about what Qt 5.15 says is needed. APKs do build in CI though so whatever setup is in there works. I have not been able to figure out how to make an App Bundle which is required to push to the store. At this point I've given up on it and not sure what to do. I just can't fight with it any more.

DonLakeFlyer, 2021-11-30:
Qt docs say JDK 11 or lower. Seems like that isn't quite correct. I can build with JDK 11 so that would seem to be the requirement.

closed 2024-06-04 by HTRamsey