mavlink/qgroundcontrol #10035: Found a bug in the QGC Airmap function.

BOB4Drone, 2021-11-24:
## Expected Behavior
If you press refresh without an AIRMAP account, a null pointer dereference occurs.

## Steps to Reproduce:

1. Turn on the QGC
2. Go to Application Settings -> AirMap -> Show Flight List
3. Press 'Refresh'

[Video](https://s3.us-west-2.amazonaws.com/secure.notion-static.com/9c45814d-dedd-4f70-ab38-e4a50cafa7a4/PX4-new_-_VMware_Workstation_2021-11-24_15-40-06.mp4?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Content-Sha256=UNSIGNED-PAYLOAD&X-Amz-Credential=AKIAT73L2G45EIPT3X45%2F20211124%2Fus-west-2%2Fs3%2Faws4_request&X-Amz-Date=20211124T065550Z&X-Amz-Expires=86400&X-Amz-Signature=e3e17da4099cf0e039899747fb2795afc46a937754f65866f22e106f1a9f3835&X-Amz-SignedHeaders=host&response-content-disposition=filename%20%3D%22PX4-new%2520-%2520VMware%2520Workstation%25202021-11-24%252015-40-06.mp4%22&x-id=GetObject)
## System Information & Build Information
- Operating System: ubuntu 20.04
- QGC Version: latest 
- Build Option: g++ -fsanitize=address

## Detailed Description

As a result of debugging with GDB, a null point dereference occurred in the function below.
https://github.com/mavlink/qgroundcontrol/blob/master/src/Airmap/AirMapSharedState.cc#L52

## Log Files and Screenshots

ASAN LOG

```
=================================================================
==254632==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x558958ec137a bp 0x7ffdb8427380 sp 0x7ffdb8426fd0 T0)
==254632==The signal is caused by a READ memory access.
==254632==Hint: address points to the zero page.
    #0 0x558958ec1379 in AirMapSharedState::login() /home/jjangu/qgroundcontrol/src/Airmap/AirMapSharedState.cc:82
    #1 0x558958ebf0aa in AirMapSharedState::doRequestWithLogin(std::function<void (QString const&)> const&) /home/jjangu/qgroundcontrol/src/Airmap/AirMapSharedState.cc:31
    #2 0x558958e83309 in AirMapFlightPlanManager::loadFlightList(QDateTime, QDateTime) /home/jjangu/qgroundcontrol/src/Airmap/AirMapFlightPlanManager.cc:850
    #3 0x55895924ff0d in AirspaceFlightPlanProvider::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) /home/jjangu/build-qgroundcontrol-Desktop_Qt_5_15_2_GCC_64bit-ASAN/moc_AirspaceFlightPlanProvider.cpp:716
    #4 0x5589592522c0 in AirspaceFlightPlanProvider::qt_metacall(QMetaObject::Call, int, void**) /home/jjangu/build-qgroundcontrol-Desktop_Qt_5_15_2_GCC_64bit-ASAN/moc_AirspaceFlightPlanProvider.cpp:888
    #5 0x55895925bfab in AirMapFlightPlanManager::qt_metacall(QMetaObject::Call, int, void**) /home/jjangu/build-qgroundcontrol-Desktop_Qt_5_15_2_GCC_64bit-ASAN/moc_AirMapFlightPlanManager.cpp:301
    #6 0x7fd5c288b348 in QQmlObjectOrGadget::metacall(QMetaObject::Call, int, void**) const (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Qml.so.5+0x2b5348)
    #7 0x7fd5c27726bc in CallMethod(QQmlObjectOrGadget const&, int, int, int, int*, QV4::ExecutionEngine*, QV4::CallData*, QMetaObject::Call) [clone .constprop.335] (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Qml.so.5+0x19c6bc)
    #8 0x7fd5c2774094 in CallPrecise(QQmlObjectOrGadget const&, QQmlPropertyData const&, QV4::ExecutionEngine*, QV4::CallData*, QMetaObject::Call) [clone .constprop.332] (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Qml.so.5+0x19e094)
    #9 0x7fd5c2774e12 in QV4::QObjectMethod::callInternal(QV4::Value const*, QV4::Value const*, int) const (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Qml.so.5+0x19ee12)
    #10 0x7fd5c2792862 in QV4::Moth::VME::interpret(QV4::CppStackFrame*, QV4::ExecutionEngine*, char const*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Qml.so.5+0x1bc862)
    #11 0x7fd5c2795721 in QV4::Moth::VME::exec(QV4::CppStackFrame*, QV4::ExecutionEngine*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Qml.so.5+0x1bf721)
    #12 0x7fd5c272a1e4 in QV4::Function::call(QV4::Value const*, QV4::Value const*, int, QV4::ExecutionContext const*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Qml.so.5+0x1541e4)
    #13 0x7fd5c28ab439 in QQmlJavaScriptExpression::evaluate(QV4::CallData*, bool*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Qml.so.5+0x2d5439)
    #14 0x7fd5c2856d4b in QQmlBoundSignalExpression::evaluate(void**) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Qml.so.5+0x280d4b)
    #15 0x7fd5c2858612 in QQmlBoundSignal_callback(QQmlNotifierEndpoint*, void**) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Qml.so.5+0x282612)
    #16 0x7fd5c288ae17 in QQmlNotifier::emitNotify(QQmlNotifierEndpoint*, void**) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Qml.so.5+0x2b4e17)
    #17 0x7fd5c283a5c5 in QQmlData::signalEmitted(QAbstractDeclarativeData*, QObject*, int, void**) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Qml.so.5+0x2645c5)
    #18 0x7fd5c1207c35 in void doActivate<false>(QObject*, int, void**) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Core.so.5+0x2d5c35)
    #19 0x7fd5bfe7d570 in QQuickAbstractButtonPrivate::handleRelease(QPointF const&) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5QuickTemplates2.so.5+0x87570)
    #20 0x7fd5bfe996f4 in QQuickControl::mouseReleaseEvent(QMouseEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5QuickTemplates2.so.5+0xa36f4)
    #21 0x7fd5c49a3f5d in QQuickItem::event(QEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Quick.so.5+0x249f5d)
    #22 0x7fd5c3c9f43b in QApplicationPrivate::notify_helper(QObject*, QEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Widgets.so.5+0x16343b)
    #23 0x7fd5c3ca5f1f in QApplication::notify(QObject*, QEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Widgets.so.5+0x169f1f)
    #24 0x7fd5c11cf807 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Core.so.5+0x29d807)
    #25 0x7fd5c49bbd66 in QQuickWindowPrivate::deliverMouseEvent(QQuickPointerMouseEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Quick.so.5+0x261d66)
    #26 0x7fd5c49bcf73 in QQuickWindowPrivate::deliverPointerEvent(QQuickPointerEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Quick.so.5+0x262f73)
    #27 0x7fd5c49bdf04 in QQuickWindowPrivate::handleMouseEvent(QMouseEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Quick.so.5+0x263f04)
    #28 0x7fd5c306f63a in QWindow::event(QEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Gui.so.5+0x17a63a)
    #29 0x7fd5c49bf3b8 in QQuickWindow::event(QEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Quick.so.5+0x2653b8)
    #30 0x7fd5c3c9f43b in QApplicationPrivate::notify_helper(QObject*, QEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Widgets.so.5+0x16343b)
    #31 0x7fd5c3ca5f1f in QApplication::notify(QObject*, QEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Widgets.so.5+0x169f1f)
    #32 0x7fd5c11cf807 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Core.so.5+0x29d807)
    #33 0x7fd5c306456c in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Gui.so.5+0x16f56c)
    #34 0x7fd5c3065954 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Gui.so.5+0x170954)
    #35 0x7fd5c30418aa in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Gui.so.5+0x14c8aa)
    #36 0x7fd5b81e5699 in xcbSourceDispatch(_GSource*, int (*)(void*), void*) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5XcbQpa.so.5+0x65699)
    #37 0x7fd5bfd1717c in g_main_context_dispatch (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x5217c)
    #38 0x7fd5bfd173ff  (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x523ff)
    #39 0x7fd5bfd174a2 in g_main_context_iteration (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x524a2)
    #40 0x7fd5c122b1cb in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Core.so.5+0x2f91cb)
    #41 0x7fd5c11ce219 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Core.so.5+0x29c219)
    #42 0x7fd5c11d71d2 in QCoreApplication::exec() (/home/jjangu/qgroundcontrol/build/staging/Qt/libs/libQt5Core.so.5+0x2a51d2)
    #43 0x558958b69d89 in main /home/jjangu/qgroundcontrol/src/main.cc:427
    #44 0x7fd5c0a190b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x270b2)
    #45 0x55895822392d in _start (/home/jjangu/qgroundcontrol/build/staging/QGroundControl+0xc6392d)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/jjangu/qgroundcontrol/src/Airmap/AirMapSharedState.cc:82 in AirMapSharedState::login()
==254632==ABORTING

```

bkueng, 2021-11-29:
@keith-bennett-airmap do you want to look into this?

keith-bennett-airmap, 2021-11-30:
Hi, @bkueng

Thanks for bringing this up. I've added an internal ticket to AirMap's ticketing system for reference, BOSS-899. I'll try to get some time to look at this soon.

Thanks,
Keith

keith-bennett-airmap, 2021-12-28:
Hi @BOB4Drone and @bkueng please see and test https://github.com/mavlink/qgroundcontrol/pull/10097

closed 2022-01-11 by bkueng