mavlink/qgroundcontrol #10036: QgroundControl not displaying Attitude/Position/Velocity PID tuning panels

dookei, 2021-11-24:
Hi there 
for some reason since I updated the px4 firmware to v1.12.3 and or Qgroundcontrol to 4.1.4 I am no longer able to see the PID tunning setup page complete. It can be that I changed something in the parameters, since I am using a custom mixer and airframe. 

This is what I get:
## Expected Behavior
[image]

This is what I had before 
## Current Behavior

[image]

## System Information
- Operating System: Ubuntu 20.04 LTS
- QGC Version: 4.1.4
- Flight Controller:  Holybro Durandal
- Autopilot : PX4 1.12.3

Any thoughts ? Am I missing any parameter to enable the Position/Velocity controller ?
Many Thanks

WTPENGUIN, 2021-11-24:
The PID Tuning page also causes problems with the Ardupilot firmware.

The page is not displayed at all.

TareqAlqutami, 2021-11-26:
I also had the same issue on Windows QGC version 4.1.4.  
I then installed the daily build and it seems to resolve the issue and all panels are back. It is also mentioned that the daily build is required for PID tuning!
https://docs.px4.io/master/en/config_mc/pid_tuning_guide_multicopter_basic.html

dookei, 2021-11-30:
Hi @TareqAlqutami thanks this worked for me.
For those who are facing the same issue. 
[https://docs.qgroundcontrol.com/master/en/releases/daily_builds.html](https://docs.qgroundcontrol.com/master/en/releases/daily_builds.html)

closed 2021-11-30 by dookei