mavlink/qgroundcontrol #10044: Support root to run QGC

dzywater, 2021-11-28:
Currently, QGC cannot be launched by root and be blocked directly by a dialog.

It can act like vscode by some args. No need to limit it fully.

booo, 2022-01-04:
I think this is on purpose and good behavior. Why do you want to launch it as root?

dzywater, 2022-02-28:
> I think this is on purpose and good behavior. Why do you want to launch it as root?
I don't know how to explain. Sometime root is more convenient, expecially for it maintenance.
Why must it be restricted to root? 
There is no need. Only add additional argvs like webots and vscode.

gromain, 2022-03-01:
Running as root if you don't need to access low level system resources is a very bad idea (and even if you need low level access to resources it's a bad idea). Leaves the door open to system exploitation.
Of course running everything as root is more convenient, but there are plenty of reasons that the root/user paradigm exists.

Please, try to explain why you need to run as root, we will try to help and find a solution.

booo, 2022-03-01:
Here is another explanation why it is bad to run things as root:

https://askubuntu.com/questions/16178/why-is-it-bad-to-log-in-as-root

Please reopen the ticket if you have good reason for your feature request. Until then I close the ticket.

closed 2022-03-01 by booo

dzywater, 2022-03-03:
I think that free choice is important and the best reason. There are always some  people to have this demand. We don't limit its scope of use.
I also compiled it myself.
[        if (getuid() == 0) {](https://github.com/mavlink/qgroundcontrol/blob/d0ea985890184fa469b373bfc1f631a909bcfb5b/src/QGCApplication.cc#L196)

mrjohnkrammer, 2022-09-21:
Hi, did you solve the problem? If so , how did you solve it

dzywater, 2022-10-04:
> Hi, did you solve the problem? If so , how did you solve it

I have to recompile it currrently.

arongu321, 2023-02-14:
Is there anyway I can run QGroundControl.AppImage inside a docker container as I'm not allowed to run it because I'm a  root user in the docker container?