mavlink/qgroundcontrol #10057: Auto-Resume Video Recording after Losing Connection

ES-Alexander, 2021-12-03:
Tell us a bit about the feature:
- _**What problem does it solve?**_
When a vehicle that's streaming video loses connection, if QGC is recording the recording stops, but doesn't automatically start recording again if/when the vehicle reconnects.
- _**What does it do?**_
It would be convenient if the recording automatically restarts when the vehicle is reconnected (perhaps within some time limit), since a momentary connection loss may not be fatal, but it can be easy to forget to resume recording if a connection has just dropped out and then resumed. It would likely be helpful to include a notification alerting the operator that recording has resumed.
- _**What flight stacks must it work with?**_
All where QGC supports local recording (at least ArduSub, not sure about others)
- _**Relevant vehicle types?**_
All where QGC supports local recording (at least submarine, not sure about others)
- _**Are there other systems that have this feature?**_
Not sure
- _**What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)**_
Recording is local, so only reliant on QGC's internal processes.

_**Any additional context you can provide will make the feature easier to evaluate (e.g. mockups, detailed specification, etc.)**_

Was brought up in [this forum post](https://discuss.bluerobotics.com/t/upper-goose-lake-12-2-21-more-trash-cleanup/11141) because the lack of this feature resulted in missing footage that was expected to be recorded. The point seemed fair, and it's something I've run into previously.

Understandably in an ideal scenario the connection would always be stable enough that this shouldn't occur, but better to expect the unexpected/undesirable than just be unpleasantly surprised when it happens.

ES-Alexander, 2022-06-01:
Would also be good to do this when switching/toggling between different camera streams.

Practically speaking the video source is changing so it makes sense that the recording stops as part of the cleanup of the previous stream (and since there's no guarantee a new stream will have the same resolution as the old one), but for users thinking about "changing view" it's unintuitive that that requires manually starting a new recording as well.

Discussed a bit in [this forum post](https://discuss.bluerobotics.com/t/how-to-toggle-cameras-in-blueos/12227/2).

koklimabc, 2024-01-18:
I also found on this issues as QGroundControl won't resume on connection after manual plug in batter power back to wifi module.

But my nvr or vrx will resume video stream connection as compared to this software.