mavlink/qgroundcontrol #10058: Add support for updated ArduPilot Compass parameter names.

IamPete1, 2021-12-03:
ArduPilot has recently changed its compass parameter naming convention. The updated names will be used for version 4.2 and later. The  change has been merged, and is now in induced in 4.2.dev.

All the per instance params have been renamed to be more consistent. The key ones for QGC are:

```
COMPASS_USE -> COMPASS1_USE 
COMPASS_USE2 -> COMPASS2_USE 
COMPASS_USE3 -> COMPASS3_USE 

COMPASS_ORIENT -> COMPASS1_ORIENT
COMPASS_ORIENT2 -> COMPASS2_ORIENT
COMPASS_ORIENT3 -> COMPASS3_ORIENT

COMPASS_EXTERNAL -> COMPASS1_EXTERN
COMPASS_EXTERN2 -> COMPASS2_EXTERN
COMPASS_EXTERN3 -> COMPASS3_EXTERN

COMPASS_ DEV_ID -> COMPASS1_DEV_ID
COMPASS_ DEV_ID2 -> COMPASS2_DEV_ID
COMPASS_ DEV_ID3 -> COMPASS3_DEV_ID
```

All the offset, diagonals, off diagonals and motor compensation parameters have also changed, but I doubt they are used by QCG.

QGC daily build currently gives a warning because it cannot find the old param names:

[image]

DonLakeFlyer, 2021-12-06:
@IamPete1  I can't seem to figure out how to get a firmware which has this in it. I tried building SITL master and testing with that. It still has old params. I flashed dev heli for testing. It still has old names. Am I doing something wrong here?

DonLakeFlyer, 2021-12-08:
@rmackay9 Am I doing something wrong here?

DonLakeFlyer, 2021-12-08:
ArduPilot PR: https://github.com/ArduPilot/ardupilot/pull/19123

DonLakeFlyer, 2021-12-08:
I went poking through the ArduPilot source on master and I'm not finding some of the changes in master which were in the pull. Now I'm even more confused.

IamPete1, 2021-12-08:
@DonLakeFlyer Sorry have been away. The changes were in master but we found a bug, so they have been temporarily reverted. I will update you once there back in tree again.

DonLakeFlyer, 2021-12-09:
@IamPete1 Do you have a pull or branch somewhere I could build my own SITL from? I'm cutting a new QGC stable literally in a couple days. I want to get this change in there. But need something to quickly test against.

IamPete1, 2021-12-09:
@DonLakeFlyer This is the PR branch, it will work fine for testing, the bug was with the param conversion coming from older firmware's. 

https://github.com/ArduPilot/ardupilot/pull/19123

Thanks for your help on this, I can build you firmware for a board if you don't have the AP dev environment setup.

closed 2025-05-03 by HTRamsey