mavlink/qgroundcontrol #10061: x track error is getting subtracted from altitude?

tstastny, 2021-12-06:
## Expected Behavior
The XTrackError fact should display the position controllers *x track error. 

## Current Behavior
Currently it always shows x track error + altitude setpoint.

## Detailed Description
The problem is here: [https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L999](https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L999)

I can make a one liner PR (remove the altitude setpoint from this calculation..), but thought I'd check quickly here to see if this was for some reason intended?

tstastny, 2021-12-06:
@AndKe it looks like you were the one to implement this in #9699. Interested in your original intent, thanks!

AndKe, 2021-12-06:
@tstastny   Nah, I have no excuse for that. most likely I just copied nearby line when I made it. Feel free to correct.  :)
Thanks

closed 2021-12-07 by tstastny