mavlink/qgroundcontrol #10070: Map view very slow

aib, 2021-12-11:
The main map view is very slow (<1 - 10 FPS) to respond to scrolling and zooming. I have a decent computer running the latest Ubuntu.

The rest of the application (i.e. Application Setup) seems fine but I can't be sure as there is UI lag of around a few hundreds of milliseconds.

## Expected Behavior
On my friend's Mac it runs fine. I would guess at least 30-60 FPS.

## Current Behavior
As stated above.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Download v4.1.4 AppImage
2. `rm -rf ~/QGroundControl* ~/.config/QGroundControl*`
3. Run the app
4. Click "OK" twice
5. Zoom and/or scroll

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 21.10
- QGC Version: 4.1.4 (AppImage prints "v4.1.3")
- QGC build: GitHub releases
- Flight Controller: none
- Autopilot (with version): none

## Detailed Description
Also tried the latest `master` (Docker build), same behavior.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
[console.log](https://github.com/mavlink/qgroundcontrol/files/7696829/console.log)
- Autopilot logs when available (post a link).
N/A
- Screenshots of QGC to help identify the current issue/bug behavior.
N/A

I could actually use some pointers for debugging this behavior in the code. Could someone point me to the most likely culprit, or at least the map widget and/or a render loop, if applicable?

adk4davsys, 2021-12-29:
Hello, I am facing the same issue with the latest appimage, have you found any solution to this issue? My map view is extremely slow to input and virtual joystick keeps giving me manual control lost error as well.

aib, 2021-12-30:
Not yet. Haven't made much progress, either. Currently lost in the Qt code.

adk4davsys, 2021-12-30:
Thanks for your reply. I also noticed when I am doing HITL with Gazebo, even opening QGC is slow. I am just going to revert back to previous version where I did not have this stuttering issue. Hope you'll figure out.

aib, 2022-01-01:
You are right, if I go back far enough the issue disappears. 4.0.11 is good whereas 4.1.4 is bad. That should give me the two commits to start dissecting. Thanks!

aib, 2022-01-01:
I have pinpointed the problem to 257f5e075d353992f925bf30ecbe37a8a0cd77f1.

remster, 2022-01-03:
I am responsible. [We considered](https://github.com/mavlink/qgroundcontrol/pull/9220) the performance implications, **saw none** and assumed there is sufficient buffering somewhere between Qt and the OS. We thus opted for simplicity. You have the original commit with the timer, please feel free to resurrect it - you clearly can repro the problem and most people cannot.

aib, 2022-01-04:
I opened #10101. @remster , I wasn't sure whether to revert #9220 and then reapply the original commit with an updated message, or do this (just add a new commit authored by myself). In the end I decided this was cleaner. Your original is almost identical to the commit I was about to make before stumbling on it, anyway.

BWStearns, 2022-03-01:
Having same issue on a maxed out Intel macbook pro with the current version available as dmg (having trouble identifying the version number).

closed 2024-05-31 by HTRamsey